mavlink/qgroundcontrol #7663: qt5AndroidDir

Ayman-git, 2019-08-05:
Hello, I tried changing the compiledsdk and buildtoolsversion to my version and it solved one of the problems that I was facing, but I found this new error that idk how to solve...

[image]

Ayman-git, 2019-08-06:
I found out that there is a error in the Mainfest and I think it might be a reason

[image]

dogmaphobic, 2019-08-06:
Please ask build questions using the [proper channels](https://app.slack.com/client/T0V5F863B/C629RE9H7)

Github issues are for bugs only. Not to mention it looks like you are using Android Studio?

closed 2019-08-06 by dogmaphobic