mavlink/qgroundcontrol #7664: Enable MAVLink Analyzer on mobile

jlecoeur, 2019-08-05:
The MAVLink Analyzer view (live plots) is available on desktop but disabled on mobile. 

It would be very useful to be able to plot live data on a tablet when on the field. What is missing/blocking to enable it on mobile?

Possibly related to: https://github.com/mavlink/qgroundcontrol/issues/3062

Jaeyoung-Lim, 2019-08-05:
This would be really useful when setting up new hardware

DonLakeFlyer, 2019-08-13:
This requires graphing library code which just isn't available on Mobile builds. So not possible.

closed 2019-08-13 by DonLakeFlyer

jlecoeur, 2019-08-13:
Wait, are you saying that it is not possible to display a graph on mobile?
Do you have a link to a previous issue, or a discussion on the subject?

DonLakeFlyer, 2019-08-13:
> Wait, are you saying that it is not possible to display a graph on mobile?

That is correct given the library we are using. QGC uses qwt libary for graphing which only supports Qt QWidget interfaces. Mobile builds are 100% native Qt Qml. As are daily builds currently which is why daily builds don't support charting either right now. Current plan for new master builds is likely to run a separate QWidget app for charting desktop builds. None of which make Mobile work. Qt Qml supports a Charting module but it is way too lame to support anything close to the current Analyze code.

jlecoeur, 2019-08-13:
OK thank you for the explanation.

I am quite surprised that Qt Qml does not have a high performance plotting module. That is a shame! Have you considered rolling out a custom charting module based on an OpenGL Qml module? 

In QGC, I spend 3/4 of the time displaying stuff on the mavlink analyzer and the rest of the time on the parameter view. Charting is my main use case, and I am sure that I am not alone.

DonLakeFlyer, 2019-08-14:
> Have you considered rolling out a custom charting module based on an OpenGL Qml module?

Not at this point because it would take a ton of new code.