mavlink/qgroundcontrol #7665: connecting keyboard(HID) to QGS (android) causes a crash

bannedas, 2019-08-05:
## Expected Behavior
Application should not crash and continue working with joystick and keyboard connected.

## Current Behavior
Application crashes when keyboard is connected

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Run QGS android version
2. Connect vehicle
3. Vehicle connected successfully
4. Connect joystick (still working)
5. Joystick recognized and working
6. Connect keyboard
7. Application not responding

## System Information
When posting bug reports, include the following information
- Operating System: Android 6.0
- QGC Version: 3.5.4
- QGC build: stable also tested daily build (same outcome)
- Flight Controller: pixhawk 4 cube
- Autopilot (with version): PX4 1.8.1

MatejFranceskin, 2019-08-10:
How do you connect flight controller, keyboard and joystick to the tablet?

Shackmeister, 2019-08-12:
Through a USB hub (actually 2 hubs)

closed 2026-10-04 by github-actions[bot]