mavlink/qgroundcontrol #7668: RTSP videostream buffer issue

Tonisvoitka, 2019-08-06:
Hi all, 

I was wondering, would it be possible to add a buffer slider for rtsp videostream? I am using a custom flightcontroller with a custom modem, that runs linux with ethernet ports, where i connect cameras and my flightcontrollers. Currently, i am using IP cameras for rtsp video stream, but the issue i am facing is that the packages are coming to the ground station in a wrong sequence. I know that this is due to the fact that the packages get lost over the LTE network. But i was wondering, would it be possible to somehow solve this on the qgc end? 

[image]

DonLakeFlyer, 2019-08-13:
Maybe @dogmaphobic or @jaxxzer would know.

Williangalvani, 2019-08-15:
I believe the [rtpjitterbuffer](https://gstreamer.freedesktop.org/documentation/rtpmanager/rtpjitterbuffer.html?gi-language=c#example-pipelines) Could help reduce this issues. I think that with a small buffer size the latency would barely be affected.

Tonisvoitka, 2019-11-28:
Is there any update on the issue?

closed 2026-10-04 by github-actions[bot]