mavlink/qgroundcontrol #7670: Display OBSTACLE_DISTANCE

mrivi, 2019-08-07:
We would like to display the vehicle distance from obstacles (mavlink message obstacle_distance) in a circular array fashion like it's done for car when you backwards.

thomasgubler, 2019-08-09:
> Plus, we would like to have an icon to show if Collision Prevention is enabled when in Position Control mode or Obstacle avoidance is enabled in Auto modes.

Clarification: this should take into account the avoidance heartbeat to give feedback to the user if the OA is running or not.

@mrivi will create a table that summarizes the various states that need to be mapped to a UX element

thomasgubler, 2019-08-12:
➤ Yasen Dimov commented:

Quick mockup of Obstacle avoidance UI:

* New indicator (top right) to show obstacle avoidance on/off
* First pass, quick mockup for obstacle distance.

thomasgubler, 2019-08-12:
➤ Yasen Dimov commented:

Alternative version for obstacle distance. However, it's huge and takes over a lot of the screen real estate. 

Happy to discuss with the people involved.

thomasgubler, 2019-08-12:
➤ Yasen Dimov commented:

Quick mockup of Obstacle avoidance UI:

* New indicator (top right) to show obstacle avoidance on/off
* First pass, quick mockup for obstacle distance.
* First pass, quick mockup for distance labels

thomasgubler, 2019-08-12:
➤ Yasen Dimov commented:

Note:  The 2nd mockup above is way too "DJI" like, but I thought something quick should be made in order to have a conversation of what we should be aiming at.

yasendimov, 2019-08-12:
UI Version 1

As stated above, a this is very quick mockup. 

[image]

Alternative version

[image]

AuterionWrikeBot, 2019-08-20:
➤ Yasen Dimov commented:

Obstacle avoidance mockup for Map View

AuterionWrikeBot, 2019-08-20:
➤ Yasen Dimov commented:

Slightly updated Camera View Obstacle avoidance visuals. 

* Now we have a more 360 degree type of view with 2 additional secorts left and right
* Obstacle avoidance indicator in the Main nav

AuterionWrikeBot, 2019-08-20:
➤ Yasen Dimov commented:

On pressing the OA indicator, we can show a small lightbox with the settings that Gus Grubba already implemented

AuterionWrikeBot, 2019-08-20:
➤ Yasen Dimov commented:

In map view, we can try to keep the visuals consisent. This is a very quick mockup, but essentially we have front, back, left and right indication of obstacles

AuterionWrikeBot, 2019-08-20:
➤ Yasen Dimov commented:

A quick breakdown of posibilities for OA on Map view

AuterionWrikeBot, 2019-08-20:
➤ Martina Rivizzigno commented:

Camera view is really nice from my point of view. As for the map view, we would need to see if the resolution is enough not to give the impression that you're flying into something when the gap between the obstacles is narrow.

valbok, 2021-07-29:
Since https://github.com/mavlink/qgroundcontrol/commit/694ac0ceb5ad65f40ad26887d17235361aedf517 has landed, can we assume this issue is closed now?

closed 2021-08-01 by DonLakeFlyer