mavlink/qgroundcontrol #7672: V3.5 android problem

zhaolianggoods, 2019-08-08:
E/libTimUQS.so(23865): ..\TimUQS\src\VideoStreaming\VideoReceiver.cc:273 (void VideoReceiver::start()): VideoReceiver::start() failed. Error with data source for gst_element_factory_make()
E/libTimUQS.so(23865): ..\TimUQS\src\VideoStreaming\VideoReceiver.cc:381 (void VideoReceiver::start()): VideoReceiver::start() failed
The creation of TCP video stream returns to NULL in gst_element_factory_make("tcpclientsrc", "tcpclient-source") during Android debugging

zhaolianggoods, 2019-08-08:
windows no problem

DonLakeFlyer, 2019-08-13:
Is this happening on your own build or one of the QGC released builds?

zhaolianggoods, 2019-08-14:
Is this happing on the QGC released builds,version3.5.0

DonLakeFlyer, 2019-08-14:
Ok, so first step is to use the latest Stable build: https://docs.qgroundcontrol.com/en/getting_started/download_and_install.html#android and verify it still happens there.

zhaolianggoods, 2019-08-15:
The latest Stable build: https://docs.qgroundcontrol.com/en/getting_started/download_and_install.html#android .there is a problem.

DonLakeFlyer, 2019-08-15:
@dogmaphobic @jaxxzer Is this just bad TCP values or some sort of bug?

Williangalvani, 2019-08-15:
This is happening in the ["gst_element_factory_make"](https://github.com/mavlink/qgroundcontrol/blob/v3.5.3/src/VideoStreaming/VideoReceiver.cc#L288), before setting anything up. I'd guess it is a bug or something is missing.

dogmaphobic, 2019-08-15:
That would imply a missing library. On Android builds, only a select few (static) libraries are included. I don't think anybody has ever tested this particular pipeline with Android before.

DonLakeFlyer, 2019-08-15:
@dogmaphobic Sentera is having problems with TCP video on iOS as well. It's not working for them on Stable. So probably same thing?

DonLakeFlyer, 2019-08-15:
Needs to be fixed in Stable.

dogmaphobic, 2019-08-15:
No, iOS uses a _Framework_. It's all included. I did not add TCP. I don't have anything that sends video this way. Video streaming within QGC is completely out of hand. Things have just been added on top of each other. RTSP for me is so broken to the point of being useless. I don't have enough experience with GStreamer to debug all these issues. I am in the process of exposing the _VideoReceiver_ class so runtime, binary plugins can implement the various video options. That's where my focus is right now.

DonLakeFlyer, 2019-08-15:
@jaxxzer Can't remember if it was your folks who added TCP support?

Williangalvani, 2019-08-15:
We only use UDP for now, it looks like it was for Sentera:
https://github.com/mavlink/qgroundcontrol/blame/14d0b71a24676bbd1fe6d2dc1c2963be0558f0c7/src/VideoStreaming/VideoReceiver.cc#L287

DonLakeFlyer, 2019-08-15:
Ok, so TCP for now is not supported on mobile. It might come back in the next Stable but maybe not.

zhaolianggoods, 2019-08-16:
Problem solving，Gstream plugin needs to be loaded

closed 2020-02-08 by DonLakeFlyer