mavlink/qgroundcontrol #7673: save px4 mission in .tlog file

moreba1, 2019-08-09:
Hi ,

I before used Arduplane code with mission planner and currently using Px4 with QGC . there is an ability in mission planner that save mission in .tlog file. can this feature added to QGC ?

DonLakeFlyer, 2019-08-13:
> there is an ability in mission planner that save mission in .tlog file

And that is used for what? I don't see how that is a useful feature.

Antiheavy, 2019-08-14:
When watching a TLOG replay you can see the onboard mission download and display onscreen in QGC.  I assume if a new mission is uploaded you see the new mission show up on screen as well?  similarly if a user "downloaded" the mission in the middle of a flight you'd see it update onscreen?

This functionality is very useful when replaying TLOG files for analysis.  You can see the intended flight path compared to the actual, etc.  

(Now if only the replay function was more useable ;-)   https://github.com/mavlink/qgroundcontrol/issues/5730 )

DonLakeFlyer, 2019-08-14:
> When watching a TLOG replay you can see the onboard mission download and display onscreen in QGC...

I've tried implementing that a couple times and gave up due to complexity. Let me see what MP does. Maybe it has a better solution for this.

@moreba1 Is there something in MP you specifically need to turn on to make this happen? If so, can you point me to it?

DonLakeFlyer, 2019-08-14:
I tried MP and it doesn't show the mission just by replaying a log either. So it must be something special.

Antiheavy, 2019-08-14:
> When watching a TLOG replay you can see the onboard mission download and display onscreen in QGC.

huh.... well with some replay logs the mission shows up for me...  and sometimes not.  I'm not sure what the pattern is.  Maybe you actually have to watch the replay until the point when a mission is uploaded?

DonLakeFlyer, 2019-08-15:
I think it might work if there is an upload in the log. But not if the mission is already on the vehicle.

moreba1, 2019-08-16:
@DonLakeFlyer 
In MP we can download missions that uploaded . this is useful in review of flight for checking quaility and for a crash investigation . we can compare flight path with mission .

[image]

DonLakeFlyer, 2019-08-16:
Ah, ok. That is different than what I was thinking it was.