mavlink/qgroundcontrol #7675: where the signal of "streamConfiguredChanged " is dealed with?

HomeLH, 2019-08-12:
I am reading the source code, I can not find the handler function of the signal of "streamConfiguredChanged".

HomeLH, 2019-08-12:
`
2 code results in mavlink/qgroundcontrol or view all results on GitHub
src/Settings/VideoSettings.h
Showing the top two matches Last indexed on Jun 27
C++
    Q_PROPERTY(bool     streamConfigured        READ streamConfigured       NOTIFY streamConfiguredChanged)
    Q_PROPERTY(QString  rtspVideoSource         READ rtspVideoSource        CONSTANT)
    static const char* videoSourceMPEGTS;

signals:
    void streamConfiguredChanged    ();

private slots:
    void _configChanged             (QVariant value);
src/Settings/VideoSettings.cc
Showing the top match Last indexed on Jun 27
C++
void VideoSettings::_configChanged(QVariant)
{
    emit streamConfiguredChanged();
}`

DonLakeFlyer, 2019-08-13:
GitHub Issues are for actual feature requests or code bugs. You can use Slack for development questions.

closed 2019-08-13 by DonLakeFlyer