mavlink/qgroundcontrol #7679: Yes/No buttons reversed from common convention in "are you sure you want to close" pop-up

Antiheavy, 2019-08-14:
The Yes and No buttons in this pop-up are reversed from common convention:
[image]

Suggest Yes on the left and No on the right.

Google image search is a wonderful reference tool:
[image]

DonLakeFlyer, 2019-08-14:
@dogmaphobic Some strangeness from the conversion to message boxes?

dogmaphobic, 2019-08-14:
This is Qt's own message dialog. They determine the order of the buttons. Even the documentation shows a negative before the positive:

[image]

https://doc.qt.io/qt-5/qml-qt-labs-platform-messagedialog.html

DonLakeFlyer, 2019-08-14:
Ick, that order is correct for OSX but incorrect for Windows.

DonLakeFlyer, 2019-08-14:
Which means Qt bug

Antiheavy, 2019-08-14:
The same warning popup is Positive then Negative in the Stable release of QGC (I'm using Windows):
[image]

dogmaphobic, 2019-08-15:
They use different versions of Qt.

DonLakeFlyer, 2019-08-15:
Stable and Daily windows use the same version of Qt. Different version of Import between Stable/Daily? Maybe go back one (or whatever) to what Stable is using? Or maybe a QGCMessageDialog which isolates the import version usage?

dogmaphobic, 2019-08-15:
Sorry, QtQuick 1.x versus QtQuick 2.x.

DonLakeFlyer, 2019-08-15:
I'm not sure about that. MessageDialog (https://doc.qt.io/qt-5/qml-qtquick-dialogs-messagedialog.html) is ```import QtQuick.Dialogs 1.3``` which isn't part of the 2.0 stuff. Stable uses 1.3, Daily uses 1.2. I bet it's just broken in 1.3.

dogmaphobic, 2019-08-15:
I tested a few variations and realized it makes no difference on macOS. Testing on something else is not that simple right now...

DonLakeFlyer, 2019-08-15:
Import 1.2/1.3 make no difference on Windows.

Antiheavy, 2019-12-17:
This seems resolved in the latest daily builds that are using newer version of Qt.  I think I can close this.
[image]

closed 2019-12-17 by Antiheavy