mavlink/qgroundcontrol #7687: Long scrollbar in console screen gets impossibly small.

edwinhayes, 2019-08-16:
## Expected Behavior
When scrolling in the console screen, a scrollbar should be visible on the RHS.  The scrollbar resizes with the amount of console output, but there should be a minimum size below which it stops getting smaller.

## Current Behavior
As the list gets longer, the scrollbar gets shorter and shorter.  Eventually, the list is so long, the scrollbar is tiny.

[image]

It's the two pixels in the image above.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Settings Page -> Console.
2. Wait a long time or otherwise generate a lot of output.
3. Scroll up and down.

## System Information
When posting bug reports, include the following information
- Operating System: macOS
- QGC build: Daily from about a week ago I think.
- Flight Controller: N/A
- Autopilot (with version): N/A

edwinhayes, 2019-08-16:
@tilaktilak

hendjoshsr71, 2023-09-05:
+1 to this 

It near impossible to scroll without an attached mouse with a wheel.

Scrolling Issues inside the Parameters Page but applies to most of the gui elements.

1. Scroll bar is non-existent
2. Up/Down arrows do not function
3. When using a mousepad the scroll speed is extremely sensitive and fast aka you cant just page to the next

closed 2026-10-04 by github-actions[bot]