mavlink/qgroundcontrol #7695: elevation tile download almost always has errors

Antiheavy, 2019-08-16:
The elevation tile downloading almost always has a handful of errors.  I suspect this could be fixed by having the map downloader more aggressively retry to download missing elevation tiles automatically.

This seems like a potential safety issue for users depending on terrain data for mission planning.  This has been happening at least since QGC 3.5.  Still happening on current daily builds.

Happens consistently with:
- multiple laptops (Windows)
- multiple internet connections 
- multiple people testing it

We have to almost always go back in and "resume download" to finish off the elevation tiles. I don't think users will actually remember to do this, especially since the elevation tile download is usually very slow.

Here are a number of screenshots showing the handful of "errors":

[image]

[image]

[image]

[image]

[image]

Antiheavy, 2019-08-16:
Here is a console log output.
[elevation-tile-download-errors_20190506-01.zip](https://github.com/mavlink/qgroundcontrol/files/3511332/elevation-tile-download-errors_20190506-01.zip)

Antiheavy, 2020-02-17:
Elevation tile downloads seem more reliable in recent Daily Builds.  Has anything changed in the last couple months or so that would make this an expected improvement?

DonLakeFlyer, 2020-02-17:
nope

Antiheavy, 2021-12-07:
I haven't noticed this issue with terrain info downloading in a long time.  I'm going to close the issue.

closed 2021-12-07 by Antiheavy