mavlink/qgroundcontrol #7700: rationalize comm in FirmwarePlugin to update APMFirmwarePlugin

tilaktilak, 2019-08-18:
I am working on compatibility with Ardupilot at the moment. 
It appears that there are several methods to send Messages from `APMFirmwarePlugin` and `PX4FirmwarePlugin` :

- [Vehicle->sendMavCommand](https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L3153) with a message queue feature. That method only allows to send CMD messages ( with 7 floats payload ).

- [Vehicle->SendMessageOnLink](https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L1932) that is basically the method used by the message queue function. That method only allows to send CMD messages ( with 7 floats payload ).

- [vehicle->missionManager()->writeArduPilotGuidedMissionItem](https://github.com/mavlink/qgroundcontrol/blob/master/src/FirmwarePlugin/APM/APMFirmwarePlugin.cc#L896) : Used to send [MissionItem](https://mavlink.io/en/services/mission.html#mavlink_commands) types.

- [guidedModeChangeAltitude](https://github.com/mavlink/qgroundcontrol/blob/master/src/FirmwarePlugin/APM/APMFirmwarePlugin.cc#L904) Create and send a set_position_target message through sendMessageOnLink. This looks like the abstraction layer (SendMavCommand) did not allow to send set_position_t* messages so it's sent directly. Note that for direct order, for instance the Goto Command, we don't want to queue messages, we prefer dropping a message than sending a too old one.

## Questions here :

- It appears that there is not send function for mavlink message [SET_POSITION_*](https://mavlink.io/en/messages/common.html#SET_POSITION_TARGET_GLOBAL_INT) or any other messages with a payload different from 7 floats.   
**It is present somewhere, or not present on purpose ?** It may be useful for instance to send [SET_HOME_POSITION](https://mavlink.io/en/messages/common.html#SET_HOME_POSITION)

- Also, It's probably good to rationalize the methods used to send messages : 
As far as I understand Vehicle is a higher level class that is both used for UI purpose and call FirmwarePlugin (lower level) to [set the flight mode](https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L2370) for instance.      
**BUT**, FirmwarePlugin use methods from Vehicle to send message, looks like a twist in the architecture hierarchy here ( It also uses several other methods from vehicle). 
Sometimes Vehicle sends directly Mavlink messages bypassing the interface class FirmwarePlugin.
Also, FirmwarePlugin calls Vehicle->MissionManager to send GuidedMissionItem, which cross 2 layers of hierarchy and create a high level method in MissionManager for a very specific case.
   
     - Was there an hierarchy in class like Vehicle->FirmwarePlugin and Vehicle->MissionManager at first ?   
     - If the answer is yes, will it make sens to : 
             - move to the common FirmwarePlugin class all the methods from Vehicle that send messages directly  ?
             - remove the call of Vehicle from FirmwarePlugin ? ( Not sure it's feasible )
             - Add a method in Vehicle to send new mavlink messages and not only the CMD ones ( with 7-floats payloads)
 
Finally, do we want another method in Vehicle to send mavlink message not queued for direct commands ? 
   
My purpose is to fully understand what's going on to update the APMFirmware and propose some cleaning if any.

Thanks !

DonLakeFlyer, 2019-08-18:
At a high level:
* Vehicle represents a vehicle. There can be multiple Vehicle objects in the system. Vehicle contains code to interact with the Vehicle which is mavlink generic and which should work against any vehicle which implements the mavlink spec. Also the Vehicle contains all the state associated with the vehicle. The majority of UI code talks directly to a Vehicle. 
* FirmwarePlugins are used to abstract code which is firmware specific into a single place and keep it from being spread through the mavlink generic mainline code. This prevents the need for all sorts of if this, if that testing within the mainline code. There is only a singleton of a FirmwarePlugin within the system associated with each firmware type and it is stateless. FirmwarePlugin code is mainly called from inside the Vehicle code as needed to implement things which are firmware specific.
* Vehicle also contains methods which are mostly abstract in that they don't work at all unless a FirmwrePlugin implements them. For these you'll see that the Vehicle code just calls the FirmwarePlugin to do the work. This creates a higher level interface to interact with a vehicle simplifying the UI. Good examples of these are all the guided command stuff like Vehicle::guidedModeTakeoff (Takeoff, RTL, GoTo Location, ...). There is no mavlink spec for any of that so the implementations of any of those commands are firmware specific. If you run QGC against a vehicle which supports mavlink but which is neither PX4 or ArduPilot these commands will be disabled.

* Vehicle::sendMavCommand is used to send COMMAND_LONG/COMMAND_INT commands to the vehicle. These commands support acking back response through COMMAND_ACK. Hence you can build retry logic to handle possibly dropped commands. This is why there is a special Vehicle::sendMavCommand implementation. It provides retry capability and the ability to notify the calling if the command failed.
* Vehicle::sendMessageOnLink can be used to send any mavlink message. Your statement that this also only sends COMMAND_LONG style 7 param messages is incorrect. You can send any mavlink message with it. It is the primary mechanism to send a mavlink message to the vehicle. Even Vehicle::sendMavCommand eventually sends the actual command to the vehicle using Vehicle::sendMessageOnLink. Except for special cases most mavlink messages are not acked. They are just fire and forget and hope it gets to the vehicle. Vehicle::sendMessageOnLink does this kind of sending. With no guarantee the message ever got to the vehicle, nor any mechanism to let you know if it failed to get to the vehicle.
* MissionManager::writeArduPilotGuidedMission item is a hack to support GoTo Location in ArduPilot. It is only used for that. It is not used to upload missions. Given the strange way this works in ArduPilot there wasn't a way to put this in the ArduPilot FirmwarePlugin which is where is really belongs along with all other firmware specific code.
* If you want to send a SET_POSITION_* message or any other random mavlink message you use Vehicle::sendMessageOnLink. There are plenty of examples throughout the code of how to build up a message and send it.

A good way to approach this would be to explain what you want to implement and I can explain how to fit that in.

DonLakeFlyer, 2019-08-18:
I think you may be confused into thinking that FirmwarePlugin is like a superclass or maybe subclass of Vehicle. That is not the case. You should think of it more like a cross platform worker library. With the cross platform aspect equating to different firmwares. It provides a consistent interface to call into with differing implementations hidden behind it.

DonLakeFlyer, 2019-08-18:
There is also AutoPilotPlugin which used to provide firmware specific vehicle setup ui implementations and expose them in the ui. All of vehicle setup (except for the parameter editor) is firmware specific. If you connect a non PX4/ArduPilot vehicle to QGC you get no Setup pages at all.

tilaktilak, 2019-08-18:
Thanks for all this explanation !
Correct, I though FirmwarePlugin was a subclass of Vehicle used to abstract low level commands, but I get it, it's kind of an add-ons of Vehicle Class to handle flight-controller specific stuff.

Basically I was first trying to fix the Goto Location that is not working with ArduPilot at the moment.
By reading the code I have seen several methods to send commands and I didn't want to came up with another one. As I've seen this writeArduPilotGuidedMission fix I was concerned how many layer of fix we have here and what was the initial idea. I'm pretty well reassured here ! 

Still don't get why ArduPilotGuidedMission has to be in MissionManager then, i have a  first fix that I came up with but I'm concerned whether or not it's the correct approach : https://github.com/Aeronavics/qgroundcontrol/tree/fix_goto

DonLakeFlyer, 2019-08-18:
> Still don't get why ArduPilotGuidedMission has to be in MissionManager then

Because the way GoTo Location is supported in ArduPilot uses the Mission Protocol and sending up a bogus WAYPOINT command as a mission to the vehicle. Since it uses the mission protocol you need to use MissionManager for that protocol implementation.

Maybe things are different now and there is a new way to do it. But that was previously the only way to implement it and get it to work across all vehicle types.

I probably should have also said that ArduPilot FirmwarePlugins are even more complicated than PX4. The reason is that not all ArduPilot vehicle type firmwares themselves even work the same way. This is why there is a Plane, Copter, Rover version of the plugin. That always brings in an extra level of complexity with ArduPilot.

tilaktilak, 2019-08-18:
You're correct, Goto is different for ArduCopter (SET_POSITION) and ArduPlane (Send a WP).

closed 2019-08-18 by tilaktilak