mavlink/qgroundcontrol #7708: Replay: Instrument cluster moves up behind toolbar

Antiheavy, 2019-08-20:
When the new replay toolbar is shown a the bottom, the instrument cluster on the right moves up and is covered by the top toolbar:
[image]

DonLakeFlyer, 2019-08-20:
@tilaktilak Can you take a quick look at this. I"m pretty sure this was working fine up until recently.

tilaktilak, 2019-08-20:
Sure, let me have a look.

tilaktilak, 2019-08-20:
I just build Commit [8a6ebbe](https://github.com/Aeronavics/qgroundcontrol/commit/8a6ebbec8dd1349371a3d096856fa67efd95a01b) before the vertical panel changes and I have the same bug, let me check if I can found what's going on.

DonLakeFlyer, 2019-08-20:
K, thanks. I can deal with it if you don't want to look.

tilaktilak, 2019-08-20:
So I've been digging in commit history and that have been introduced by commit [fc8fd6a](https://github.com/Aeronavics/qgroundcontrol/commit/fc8fd6a)

DonLakeFlyer, 2019-08-20:
So it was there all long. Strange I never noticed. I'll fix it.

DonLakeFlyer, 2019-08-20:
@tilaktilak Thanks for taking a look.

tilaktilak, 2019-08-20:
Maybe not related, but I've also found some bug in MultiVehichle panel (left side of the panel is cropped)  that could be addressed at the same time
Had a quick look and I think it's because the width is fixed to something that could be smaller that the text displayed : https://github.com/mavlink/qgroundcontrol/blob/master/src/FlightDisplay/FlightDisplayView.qml#L544

DonLakeFlyer, 2019-08-21:
Can you put details for the multi-vehicle thing in here as well and I'll take care of them togther.

tilaktilak, 2019-08-21:
Ok, there might be 2 bugs here, 

I addressed one of them in my [PR](https://github.com/mavlink/qgroundcontrol/pull/7613/) : 

- The selector between single/multi is overlayed by the instrument panel.   

     See Here : https://github.com/mavlink/qgroundcontrol/pull/7613/commits/aa6fb469ec1e2d57469e42ef66c9f2a7aa88f2d2
    And do not hesitate to review if my fix is alright :smile: 

- Part of [`MultiVehicleList`](https://github.com/mavlink/qgroundcontrol/blob/master/src/FlightDisplay/FlightDisplayView.qml#L544) is cropped here.

[image]

closed 2019-09-03 by DonLakeFlyer