mavlink/qgroundcontrol #7709: Replay:  live scrubbing does not follow cursor very well

Antiheavy, 2019-08-20:
Using Windows.  This has been happening for a while, but is way worse in the latest daily builds.   when trying to live scrub the replay timeline, if you move the cursor halfway across the timeline, the marker only moves a very small percentage of the distance the cursor was dragged. clicking at a spot on the timeline just causes it to reset to Zero.

Antiheavy, 2019-08-20:
I just noticed that this problem seems to match up to time gaps in the TLOG, e.g. in a lost comms situation or a large period of time that the app was left on after a loss of comms (e.g. landing and pulling the battery)

DonLakeFlyer, 2019-09-06:
Played around with this a bit. I think changing away from time being to range for the slider is a mistake. Percentage of bytes is just too meaningless. Depending on how your telemetry rates are set up you end up with what seems like random movement through the log. I think only time is really going to make sense to a user.

> It seems like TLOGs with large time gaps are bound to be common. Most commonly because users don't close down the app or disconnect the vehicle after landing.

I'm not sure why this would be common. The log is closed when:
* app shuts down
* last vehicle goes away
* vehicle disarms

Antiheavy, 2019-09-06:
App shuts down:  users very often just leave it up and running, for days sometimes!

Last vehicle goes away: users generally pull the battery out of the vehicle or shut it off. QGC sees this as a loss of Comms event and maintains the connection open waiting to reconnect. I don’t know that I’ve ever seen a user press the “disconnect” button, because why would they?

Vehicle disarms: in our use case, for various reasons,  the vehicle often does not disarm after landing

DonLakeFlyer, 2019-09-06:
> I don’t know that I’ve ever seen a user press the “disconnect” button, because why would they?

That's somewhat badness. It means if they reconnect the vehicle, it is running from old state since it doesn't re-query information.

DonLakeFlyer, 2019-12-24:
Still not sure how to make this better. Would require parse of entire file to find gaps to make the scrubbing understand there was gaps.