mavlink/qgroundcontrol #7738: Current WP is movable

tilaktilak, 2019-08-30:
## Current Behavior
While in FlyView the current WP is now green and surrounded by a white circle. It's also movable when we click on it even in FlyView (without updating the flight plan).

Probably since : Plan View Work #7721 
@dogmaphobic if you can confirm ?

## Steps to Reproduce:
1. Create a flight plan and fly it
2. In FlyView, click on the green WP to move it

## Version tested 
 65a7c5c117a56fcbca3e084f6b9b027bdeccc6a2

## Log Files and Screenshots
[image]

dogmaphobic, 2019-08-30:
It's always been movable. All I did was to add a circle around the currently selected waypoint. 
@DonLakeFlyer Are waypoints supposed to be movable (drag & drop)?

DonLakeFlyer, 2019-08-30:
> It's always been movable.

Hah, funny. It's been like that for years then. It's not supposed to be movable.

DonLakeFlyer, 2019-08-30:
> It's been like that for years then.

Not years, but broke somewhere along the way. It's supposed to be off:
```
    Component.onCompleted: {
        showItemVisuals()
        if (_missionItem.isCurrentItem && map.planView) {
            showDragArea()
        }
    }
```

Must have lost contact with map.planView somewhere.

DonLakeFlyer, 2019-08-30:
> Must have lost contact with map.planView somewhere.

Nope again! Back to been there for years. Missing check for planView on dynamic update:
```
    Connections {
        target: _missionItem

        onIsCurrentItemChanged: {
            if (_missionItem.isCurrentItem) {
                showDragArea()
            } else {
                hideDragArea()
            }
        }
    }
```

closed 2019-08-30 by DonLakeFlyer