mavlink/qgroundcontrol #7743: Segmentation fault on start with latest Daily (Ubuntu 18.04)

MaEtUgR, 2019-08-31:
## Expected Behavior
QGC starts normally like the older daily versions I was running before.

## Current Behavior
The latest daily I downloaded today (31.08.2019) segfaults immediately on start before the window comes up:
[image]

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. `wget https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/master/QGroundControl.AppImage`
1. `chmod +x QGroundControl.AppImage`
1. `./QGroundControl.AppImage`
1. See segfault

## System Information
- Operating System: Ubuntu 18.04 in VMWare on Windows 10 host
- QGC Version: Latest Daily 31.08.2019 16:50 CET, I have no idea how to extract the version from the binary if it doesn't start.

## Detailed Description
I did not attach any autopilot via USB or UDP. Older versions are still working fine e.g. the last I used on that computer was the daily build based on aa2c794c0 (quite old).

dogmaphobic, 2019-08-31:
The error is shown in the console as soon as you launch it:
```
$ ./QGroundControl.AppImage 
/tmp/.mount_u6DmxD/QGroundControl: error while loading shared libraries: libcrypto.so.1.0.0: cannot open shared object file: No such file or directory
```

That is a missing OpenSSL dependency. @MatejFranceskin

MaEtUgR, 2019-08-31:
@dogmaphobic Thanks for the quick check. Why do I not have the same error as you mention in your comment? I have OpenSSL installed. Here's the text out of the screenshot above:
```
maetugr@ubuntuvm:~$ ./QGroundControl.AppImage
Settings location "/home/maetugr/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "*Log.debug=false\n"
GStreamer debug output: "/home/maetugr/Documents/QGroundControl/Logs/gst/gstreamer-log.txt"
Error loading text-to-speech plug-in "speechd"
serialnmea: No known GPS device found. Specify the COM port via QT_NMEA_SERIAL_PORT.
Failed to set Geoclue positioning requirements. Geoclue error: org.qtproject.QtDBus.Error.InvalidObjectPath
QGCPositionManager error 2
MAVLinkLogManagerLog: MAVLink logs directory: "/home/maetugr/Documents/QGroundControl/Logs"
System reported locale: QLocale(English, Latin, UnitedStates) "en_US"
Loading localization for "en_US"
Qt localization for "en_US" is not present
Error loading application localization for "en_US"
Map Cache in: "/home/maetugr/.cache/QGCMapCache300" / "qgcMapCache.db"
Segmentation fault (core dumped)
```

dogmaphobic, 2019-08-31:
Ha! We have different issues. Mine died earlier because of OpenSSL. Yours went a lot further. Let me see if I can duplicate (I was running on 19.04. Let me boot 18.04)

dogmaphobic, 2019-08-31:
Good question. I got the same crash as yours when running on 18.04. However, if I build it from sources (same as the one used to generate the app image above), it runs fine. Something messed up in generating the app image?

dogmaphobic, 2019-08-31:
I don't know. I built a release build and run `create_linux_appimage.sh` manually and the resulting app image had no issues. I could not find in the build when it is generated. @DonLakeFlyer? In looking at the Travis build logs, I could not find where the app image is created.

DonLakeFlyer, 2019-08-31:
> I could not find in the build when it is generated.

https://github.com/mavlink/qgroundcontrol/blob/master/.travis.yml#L213

There are forum reports on this failing on I think it was ArchLinux as well due to wrong libcrypto

DonLakeFlyer, 2019-08-31:
AppImage builds from travis need to be verifed on a virgin Ubuntu install to verify that they really work.

dogmaphobic, 2019-08-31:
I was looking into the Travis logs. I now found it in the latest build (I could not find in some random earlier build): https://travis-ci.org/mavlink/qgroundcontrol/jobs/579108795

I don't see anything extraordinary about the log output. Running the script here produces a functional appImage. I get the same crash as Matthias when running that particular appImage. This is not related to missing OpenSSL. If you have the missing OpenSLL issue, it happens immediately and it says so on the console.

MaEtUgR, 2019-08-31:
@dogmaphobic Same as yours on my fresh Ubuntu 19.04:
```
user@ubuntu1904:~$ ./QGroundControl.AppImage 
/tmp/.mount_JWgqWb/QGroundControl: error while loading shared libraries: libcrypto.so.1.0.0: cannot open shared object file: No such file or directory
```
It seems the OpenSSL library `libssl1.0.0` is not packaged for newer distros anymore or at least I can't find it:
```
user@ubuntu1904:~$ find / -name "libcrypto.so*"
/usr/lib/x86_64-linux-gnu/libcrypto.so
/usr/lib/x86_64-linux-gnu/libcrypto.so.1.1
```
It's not listed anymore for > Ubuntu 18.10 https://packages.ubuntu.com/cosmic/libssl1.0.0

dogmaphobic, 2019-08-31:
Yes, we're talking about two separate issues. One is the dependency of a specific version of OpenSSL. That's something that is showing up on Ubuntu 19.01 and some other distributions. @MatejFranceskin added some code that requires linking against OpenSSL and that's the root of that issue. OpenSSL was already used by other parts of QGC. @DonLakeFlyer would know better what that is. We need to coordinate the use of OpenSSL to avoid depending on a specific version.

The other problem, the one you created this Github issue for is something else. The app image is crashing and I can't find out why. When building the image locally, using the same code base (but likely a very different build environment), it works fine.

DonLakeFlyer, 2019-08-31:
The previously existing OpenSSL stuff was for Qt to use so you can download from https: urls. QGS doesn't really use directly just indirectly through Qt.

DonLakeFlyer, 2019-09-18:
More and more compaints on this. 

@dogmaphobic @stefandunca Paring work broke master almost three weeks ago. This really needs to get fixed. This sort of stuff just can't sit around for so long unfixed!

dogmaphobic, 2019-09-18:
This stuff should not be on master unless it's been done and tested. All this work has been on a intermediate branch. I'm not sure what is going on here. @MatejFranceskin?

DonLakeFlyer, 2019-09-18:
https://github.com/mavlink/qgroundcontrol/pull/7718

DonLakeFlyer, 2019-09-18:
Anything changes related to OpenSSL changes needs to be verified to work correctly on all OS:
* Make sure it runs on a virgin machine
* Make sure both Firmware downloads and Terrain still works both of which requires OpenSSL support

stefandunca, 2019-09-18:
The OpenSSL dependency seems unrelated to the original topic.

@DonLakeFlyer `PairingManager` needs `OpenSSL` hence the linking dependency. AFAIK this was always a requirement of QGC for linux due to QT networking and it's usage in tile fetching for some sources.
It seems to me that the root cause of the issue is that the`AppImage` is linked with `OpenSSL 1.0.0` and the newer systems are shipped with only `OpenSSL 1.1.x` by default. Installing `OpenSSL 1.1.1` should fix the issue. @MatejFranceskin implemented PairingManager to work with any of the two OpenSSL versions because the newer one (1.1.x) is not backward compatible with 1.0.0.
Regarding QT requirements:
 - until `5.11` network module was linking with OpenSSL 1.0
 - from `5.12` onwards QT links with OpenSSL 1.1.x
I already started investigating and the following days I will look into integrating OpenSSL 1.1.1c libraries into the Android APK and linux AppImage which should fix all this dependencies.

DonLakeFlyer, 2019-09-18:
Ok, so other than this strange crash. The other reports were from newer OS right which won't work with Qt 5.11 anyway right?

DonLakeFlyer, 2019-09-19:
On a virgin Ubuntu 18.04 the QGC daily build will crash on startup.

DonLakeFlyer, 2019-09-19:
Also travis builds the released AppImage on 16.04 not 18.

DonLakeFlyer, 2019-09-19:
Can someone:
* Turn off pairing in master in a pull
* Let it flow through CI, merge in and test the generated AppImage daily on a virgin Ubuntu 18.04
* Then you can put pairing back on

It seems curious to me that the changes for pairing which include new libs and code goes in and then a few days later we get reports of crashes.

MaEtUgR, 2019-09-19:
If you have such an AppImage you can always share it with me for testing...

MaEtUgR, 2019-10-03:
I just tested again today, fixed 👍

closed 2019-10-03 by MaEtUgR

MaEtUgR, 2019-10-03:
btw also on 19.04 so the other problem we discussed is gone as well.