mavlink/qgroundcontrol #7751: Segfault on TOGA joystick calibration

LorenzMeier, 2019-09-03:
@dogmaphobic When calibrating the joystick on TOGA QGC crashes when the calibration is done (all four axes) and you try to click next.

dogmaphobic, 2019-09-03:
That's because the WIP code for gimbal calibration, which was disabled was set to enable without the proper code to support it.

AuterionWrikeBot, 2019-09-03:
➤ Gus Grubba commented:

From Slack:

Who enabled gimbal in the joystick configuration? That code was not done:
https://github.com/mavlink/qgroundcontrol/blob/qgc-centos/src/VehicleSetup/JoystickConfigController.cc#L17

stefan  26 minutes ago
I enable it in order to test the joystick control for gimbal. What exactly is still missing?

stefan  22 minutes ago
@gus might this affect the normal joystick behaviour?

gus  20 minutes ago
Yes, that wacky "state machine", which defines what to do next with all that hard-coded stuff is not done. Everything in there is hardcoded to specific channels and behaviors, which only work for the 4 attitude channels. I started working on that and got utterly confused by it. I left the WIP code in there but disabled.

gus  19 minutes ago
None of these joysticks have more than 4 analog channels so there was no rush in trying to sort out that code. Just put it back to disabled and the crash will go away. (edited)

gus  17 minutes ago
The "state machine":
https://github.com/mavlink/qgroundcontrol/blob/cento/src/VehicleSetup/JoystickConfigController.cc#L145

dogmaphobic, 2019-09-03:
Fixed in https://github.com/mavlink/qgroundcontrol/pull/7759

closed 2019-09-03 by dogmaphobic