mavlink/qgroundcontrol #7760: Fences shown on Flight Data screen but not on Flight Planner screen

rmackay9, 2019-09-04:
The next versions of ArduPilot Copter and Rover will support multiple inclusion and exclusion polygons and circles but I've noticed while testing with QGC that the Flight Planner screen (perhaps there's another name for the screen that users use to create waypoints) isn't correctly showing fences that it knows about if the flight controller has been rebooted.

The screen shots show that QGC knows about two inclusion polygons (they appear on the flight data screen) but they don't appear in the list of fences on the flight planner screen.
[image]

[image]

rmackay9, 2019-09-04:
This is using a very recent version of QGC "latest" from a week ago or so.

DonLakeFlyer, 2019-09-04:
Must be some specific sequence problem. I tried this on master:
* Connect to PX4 SITL
* Add fence from Plan View, Upload
* Back to Fly View
* Kill SITL
* Communication Lost, Disconnect
* Restart SITL
* QGC connects, Fly view shows fence
* Switch to Plan View, shows fence

@rmackay9 Can you give me a specific set up steps?

DonLakeFlyer, 2019-09-04:
Could also be ArduPilot specific but I kind of doubt that.

rmackay9, 2019-09-04:
@DonLakeFlyer, ah, I guess my wording wasn't very clear at all.  It's actually not restarting the flight controller that causes the issue, it's restarting QGC.

- start up the flight controller
- in QGC connect and create a fence and upload to the flight controller
- restart QGC and you should see the issue

rmackay9, 2019-09-04:
.. as a side note, on the flight planner screen the "Rally" label which is next to the "Fence" selection button (and below the Upload button) isn't visible.  This little issue is also visible in the screen shot above.

DonLakeFlyer, 2019-09-05:
> as a side note, on the flight planner screen the "Rally" label which is next to the "Fence" selection button (and below the Upload button) isn't visible. This little issue is also visible in the screen shot above.

If you run latest master the ui for this is changed and should work now.

DonLakeFlyer, 2019-09-05:
> It's actually not restarting the flight controller that causes the issue, it's restarting QGC.

@rmackay9 Perfect, thanks. I can repro.

closed 2019-09-05 by DonLakeFlyer

rmackay9, 2019-09-06:
yes, working now, thanks!

PR #7768 (2019-09-05): Fix initial load of fence/rally for Plan view
Fix for #7760.

The bug/complexity here relates to this case:
* User creates a Plan while offline
* User connects to vehicle
* We don't want to blow away the Plan the user created while offline with the Plan from the vehicle

The fix I made makes this sequence better in that it fixes #7760. But still not fully correct. There are still problems if the Plan the user created say has no fence, but the vehicle does have a fence in the scenario above.