mavlink/qgroundcontrol #7770: Remaining work items for completion of next Stable

DonLakeFlyer, 2019-09-05:
* Implement Set RC to Param in QML
* Allow Mavlink Inspector to float in seperate window
* Move all builds to Qt 5.12.4
* Move windows to 64 bit only and Visual Studio 2017
* Move terrain backend to pulling from a directory of height (zipped .hgt format) file instead of airmap
* Bring back Analyze Widget Graphing
* Figure out how to get OpenSSL working in android 64 bit version. Currently airmap terrain and pairing are disabled.
* Update travis build for OSX to sign builds. OSX Catalina is going to prevent currently unsigned builds from running at all.

Extra Credit:
* Automate iOS release builds upload to Apple Store. So Gus doesn't have to do this manually everything.

DonLakeFlyer, 2019-09-05:
* Window size/position is no longer preserved as a setting. This used to be in MainWindow.cc. I think it's just gone now. Mean you can't reszise the window and how it show up that size next boot.

DonLakeFlyer, 2019-09-05:
* Plan/Fly buttons are duplicated in the Toolstrip. They need to be removed.

DonLakeFlyer, 2019-09-07:
* Mavlink inspector does not support multi-vehicle

DonLakeFlyer, 2019-09-07:
> * Window size/position is no longer preserved as a setting.

This one should be prioritized first. It makes testing things on master a royal PITA.

DonLakeFlyer, 2019-09-07:
I merged all the comments into a single list and split into missing feature and maintenance.

Antiheavy, 2019-09-08:
> Plan/Fly buttons are duplicated in the Toolstrip. They need to be removed.

You are talking about getting rid of these correct?  If so, hooray!
[image]
[image]

DonLakeFlyer, 2019-09-09:
> You are talking about getting rid of these correct?

Nope. The ones in the toolbar stay. The ones in the toolstrip on the left edge go away. Just like it was before.

Antiheavy, 2019-09-09:
> The ones in the toolstrip on the left edge go away. Just like it was before.

Ah.  Too bad.  IMHO having them on the left-side toolbar is more intuitive and much better overall.  The look and feel could be adjusted, but switching between Fly and Plan views should feel like you are in the same "screen", but just switching context. Both Fly and Plan are the views that are used every time and should feel like you are interacting with the aircraft.  The top toolbar buttons like App settings, Vehicle setup, and Analyze are very different in they take you to screens "away" from the normal day-to-day interaction with the aircraft.

Anyway, I suppose there is risk of mutiny if you were to take away Plan and Fly from the top toolbar, but whoever moved them down to the left-side toolbar had the right idea IMHO.

DonLakeFlyer, 2019-09-09:
> Anyway, I suppose there is risk of mutiny if you were to take away Plan and Fly from the top toolbar, but whoever moved them down to the left-side toolbar had the right idea IMHO.

They were duplicated temporarily to solve a custom build case where there was no toolbar items at all. Hence they needed to be somewhere. Not to solve the Fly/Plan really need to merge problem. That's a whole other bucket of complications.

DonLakeFlyer, 2019-09-28:
@thomasgubler Can I get some sort of timeline from Auterion on fixing the first set of problems? A new Stable can't go out without these being fixed.

bkueng, 2019-12-04:
The 'onboard files' dialog is also missing.

mrpollo, 2019-12-05:
@DonLakeFlyer is this still an accurate representation of what's missing before the next stable release?

DonLakeFlyer, 2019-12-10:
Yup

DonLakeFlyer, 2019-12-11:
> * Move terrain backend to pulling from a directory of height (zipped .hgt format) file instead of airmap

I bailed out on this one. After doing a pile of work against it I realized I had on 3 arc-second data whereas in some places it looks like airpmap has 1 arc second data. So it ended up being a downgrade. I still have the pull and will get back to some variant of in the future.

DonLakeFlyer, 2019-12-13:
Fly/Plan in toolstrip removed here: https://github.com/mavlink/qgroundcontrol/pull/8121

DonLakeFlyer, 2019-12-13:
I'm going to release with these features missing:
* [ ] Set RC to Param feature is missing
* [ ] Allow Mavlink Inspector to float in seperate window
* [ ] Analyze Widget Graphing feature is missing

With details in the release notes as to why. I've waited 6+ months for them to be put back. Time is up.

DonLakeFlyer, 2019-12-22:
Android OpenSSL stuff can come from here: https://github.com/Auterion/android_openssl.git

thomasgubler, 2020-01-06:
I think "Analyze Widget Graphing feature is missing" is done through https://github.com/mavlink/qgroundcontrol/pull/8199

DonLakeFlyer, 2020-02-08:
@dogmaphobic The only thing left here is Set RC to Param. Not sure what that status of that is. If done. You can close this whole thing out.

DonLakeFlyer, 2020-02-21:
Closing this in favor of detailed individual issues

closed 2020-02-21 by DonLakeFlyer