mavlink/qgroundcontrol #7773: Planned Home position ideas

tilaktilak, 2019-09-05:
This issue is a follow-up to the discussion from https://github.com/mavlink/qgroundcontrol/issues/7765

We want to open the discussion to address several things :
- No Takeoff is created while we start our mission edition by a complex item
- "Planned Home" automatically created by WP is fine but it could be confusing when it's put at the corner or a survey or the beginning of any complex scheme. For instance you create your survey that shows up as a standard rectangle on the map, then you size it and the planned home ends up either far away or in the middle of it.

The idea here is to explicitly ask the user to add the Home as the first item of the mission, then we can also attach a Takeoff item (which has to be non-removable as PlannedHome is) to this position.

I am working on adding a "Home" button to the ToolStrip to create the PlannedHome
The idea is to hide/disable the other tools but Home at first and when the PlannedHome is set show all other tools but Home.

[image]

Antiheavy, 2019-09-08:
Is this envisioned for Ardupilot or PX4 or both?  There are many use cases where a Takeoff waypoint is not required, e.g. when taking off in another mode and then switching to Mission mode after in the air.

Also, Fixed Wing takeoff waypoints typically are placed far out in front of you where you will climb out to (at least in PX4 that is how it works).

I'm also a less familiar with Ardupilot, but with PX4, it is the vehicle's actual home position that matters, the "planned home" is just a reference for planning when not connected to a vehicle.  Qgroundcontrol automatically places a planned home position, so this proposed change is in some ways adding a step instead of saving a step.

tilaktilak, 2019-09-08:
Thinking about this, the main issue is to insure that the first missionItem is a Takeoff in the case of Ardupilot ( Copter only ? ). This could be addressed separately.

Other than that, I think the "planned home" is more a reference for both PX4 and Ardupilot. But if this reference is mis-placed automatically every time I think it make sens to either :
- Place it explicitly at the beginning of the mission, then the telemetry distance/ distance / time will make way more sens than if the planned home is automatically set on the opposite corner of the survey we are planning.
- Remove the "Planned Home" label if no-one use this point to define their mission and keep this "planned home" in backend to process the flight plan stats.

DonLakeFlyer, 2019-09-09:
The issues with Planned Home position and Takeoff insertion should be treated as separate issues. They aren't really related to each other. @tilaktilak Could you separate these out?

Antiheavy, 2019-09-09:
> and Takeoff insertion should be treated as separate issues.

One thought I had was possibly an application setting option for "prompt user for Takeoff WP" and related "prompt user for Landing Pattern".  That way users (or custom builds) could choose to be "helped" with those items or not.  The details of the "help" are probably up for hot debate.

tilaktilak, 2019-09-09:
> @tilaktilak Could you separate these out?
I agree, let me spit them.

> One thought I had was possibly an application setting option for "prompt user for Takeoff WP" and related "prompt user for Landing Pattern". That way users (or custom builds) could choose to be "helped" with those items or not. The details of the "help" are probably up for hot debate.

I think it's kind of firmware dependent and as soon as we are suppose to select the Firmware we want to use ( it's most of the time the previously used one) QGC can know if a TakeOff and a Landing pattern is needed.

Antiheavy, 2019-09-10:
> I think it's kind of firmware dependent and as soon as we are suppose to select the Firmware we want to use ( it's most of the time the previously used one) QGC can know if a TakeOff and a Landing pattern is needed.

Good point.

tilaktilak, 2019-12-10:
#7799 is enough to fix the initial issue and the proposed implementation is "Ardupilot" only so not relevant.

closed 2019-12-10 by tilaktilak