mavlink/qgroundcontrol #7774: Terrain Following Fix and Improvements

tilaktilak, 2019-09-05:
This issue is a follow-up to the discussion from #7765 

 ## Terrain Altitude of Planned Home is not determined properly
At first it's not determined, but works if we move it. Here the error shown :   
```
TerrainQueryLog: Error in fetching elevation tile. Empty response.
```
[image]

TODO :

- Currently QGC does not support ArduPilot vehicle terrain requests where terrain maps are uploaded to the vehicle such that things like an RTL and follow terrain

- Changing to a new terrain server and supporting vehicle terrain requests. (Probably a big item)

DonLakeFlyer, 2019-09-06:
So the first phase for this is to switch to a new terrain server. I want to do that like ArduPilot does which is using height files (.hgt, https://dds.cr.usgs.gov/srtm/version2_1/Documentation/SRTM_Topo.pdf). Auterion has agreed to host the server. ArduPilot folks are ok with QGC hitting there servers until Auterion can stand up the data.

I had already done some false starts at coding this. The trickiest part is going to be the fact that normally height files are zipped. Which means QGC needs cross-platform unzip capability. That likely means including the zlib source in QGC and building it ourselves. Would could create a different format for zipping which would work with Qt quncompress but having unzip as a general capability in QGC is useful for more than just terrain. If we had unzip we could read .kmz files instead of just .kml files.

Once we get the file downloaded (using QGCDownload), and unzipped. It's just a matter of translating into the binary format for our offilne database. TerrainQuery.* and TerrainTile.* are a good place to start looking at that.

tilaktilak, 2019-09-08:
That's great for #7790 ! 
I can have a look at zlib and how to compile it then.

tilaktilak, 2019-09-09:
I notice that zlib looks present already in our repo : https://github.com/mavlink/qgroundcontrol/tree/master/libs/zlib/windows

edwinhayes, 2019-09-09:
The implementation should to be able to support DEM tiles from a GIS tile server in GeoTIFF format (for example).  What is the reason for using the SRTM height format in the first place?

DonLakeFlyer, 2019-09-09:
> What is the reason for using the SRTM height format in the first place?

Because it's a compact format which compresses well. Which in turn means download times are fast.

DonLakeFlyer, 2019-09-09:
> I notice that zlib looks present already in our repo, at least for windows :

Ah, interesting. Didn't know that. @dogmaphobic What's going on with this zlib? Can we get support for all OS?

tilaktilak, 2019-09-09:
> > What is the reason for using the SRTM height format in the first place?
> 
> Because it's a compact format which compresses well. Which in turn means download times are fast.

I think the question was why not using a tile server software like OpenMaptiles or GeoServer. Then we can query tiles as it's done by QGC Map engine (with x,y and zoom). The transfer speed optimization is handled by the tile service itself. This could also allow to overlay terrain in a map for instance.

edwinhayes, 2019-09-09:
> I think the question was why not using a tile server software like OpenMaptiles or GeoServer. Then we can query tiles as it's done by QGC Map engine (with x,y and zoom). The transfer speed [...]

Correct.  Also, it means the existing functionality for precaching tiles should be able to be reused for precaching the terrain data, which is important since Ardupilot can ask for tiles in flight which weren't part of the original mission: http://ardupilot.org/copter/docs/terrain-following.html

DonLakeFlyer, 2019-09-09:
QGC already has terrain height caching support. The format of the data doesn't really make that any easier/harder.

The problem with existing terrain height servers is that they tend to come with all sorts of varying license agreements. The one killer that always seems to show up is not being able to cache the data locally. Other one is support for hitting it from all around the world. Hence not fighing with license agreements tends to be easier.

> This could also allow to overlay terrain in a map for instance.

That would require a lot of additional work on map display. Seems unrelated to terrain height discussion.

edwinhayes, 2019-09-09:
> QGC already has terrain height caching support. The format of the data doesn't really make that any easier/harder.

That's not necessarily true; the SRTM data only provides for elevation data at a fixed resolution, whereas other formats support regionation.  Does QGC's caching model support regionation of data to cope with caching extremely large areas?

> That would require a lot of additional work on map display

It's work which [we're] going to have to do, regardless of whether it gets added to the mainline.  Mission Planner supports it today and our customers want it.

DonLakeFlyer, 2019-09-09:
I think QGC cached data is fixed at something like 30m spacing.

> Does QGC's caching model support regionation of data to cope with caching extremely large areas?

Not quite sure what that means. But if it means, zoom levels on the terrain height data so you can store large areas at low zoom level then I get it. And yes that would be much better than whats there now. Wasn't thinking along those lines.

edwinhayes, 2019-09-09:
Sorry, yeah that's right: replacing the high resolution data (in each 'region' I guess) with a single lower resolution value.

tilaktilak, 2019-09-10:
Having a look at the software I realized that Elevation tiles seems to be downloadable through Offline Maps menu. Does someone knows how it works and where this elevation data is used ?

[image]

tilaktilak, 2019-09-10:
for reference : https://github.com/mavlink/qgroundcontrol/pull/5842

DonLakeFlyer, 2019-09-10:
* You can set the Altitude on items to be "Above Terrain". This will set the actual altitude of the item to be an AMSL value being terrain height at that point + altitude.
* Survey/Corridor support a Terrain mode which you can turn on. This will generate a survey which which is a certain amount of height above terrain. It will create interstitial items within the transect to keep it at that height based on a tolerance and max/climb descent rates. Create a survey, turn it on, load to vehicle. Pull it back from vehicle if you want to see what it actually generated.
* Click the "T" next to +/- and it will show the mission status which will show terrain heights for each waypoint in relationship to waypoint height. It will turn the bar red if the item is below terrain.

DonLakeFlyer, 2019-09-10:
Example terrain waypoints:
[image]

Terrain follow survey:
[image]

tilaktilak, 2019-09-10:
Thanks for all this info,
So can the elevation tiles be cached offline (same as maps) ?

DonLakeFlyer, 2019-09-10:
> So can the elevation tiles be cached offline (same as maps) ?

yup

tilaktilak, 2019-10-07:
Following discussion here : #7864

@cosmarc 
So I rework a little bit the TerrainQuery code + create a generic class for MapProvider ( Then a big hash table contains all providers ). 
This is beneficial for both map/elevation. We can create a child of MapProvider that will fetch data from a WMS server, or even a SRTM mirror, same for 
I ended up adding gdal to QGC to open geotiff and apply transformation, so with gdal onboard theorically we can open every format compatible ( See here : https://gdal.org/api/raster_c_api.html#_CPPv48GDALOpenPKc10GDALAccess and here : https://gdal.org/drivers/raster/index.html) ! 
On my current version QGC search in a folder for .tiff files to try to retrieve elevation data from it, if it's not available fallback to Airmap elevation.
My current working branch is really messy (include some UI changes on OfflineMap that are just random experiments to connect things together) but it's here : https://github.com/Aeronavics/qgroundcontrol/tree/MapProviders_step2
I Also had in mind adding the ability to manage several elevation provider ( and class them by priority ) and same thing with the map provider (Adding the ability to overlay maps and put an orthophoto on the top of a basemap as it appear useful for our customers, that is basically the purpose of my messy UI changes in MapProviders_step2 branch).

There is a first PR related to the creation of a generic class for MapProviders here : https://github.com/mavlink/qgroundcontrol/pull/7818

My work/challenges at the moment : 
- Link gdal statically to optimize QGC binary size.
- I manage to compile my code on Windows, Linux and Android with dynamic libs but have no access to an OSX platform.
- I need a nice UI menu to connect this backend ( How to add  elevation providers map layers from a given list, add custom element in this list (with a WMS url or a local folder of tile), handle layers ordering, opacity for basemap. Integrate the ability to overlay maps in FlightMap item ( I though about a repeater of FlightMap items somewhere between the current map and the items on the map ).

cosmarc, 2019-10-08:
@tilaktilak  Nice work!
I've managed to statically compile libproj (6.2.0) and gdal (3.0.1) on linux but the size of the binary is huge (93MB with most of the drivers disabled). Let's have a call on skype(ID: cosmin_marc) and discuss more. I think we can work together, especially on the backend part.
I have a few questions:
- Why did you choose gdal 2.4.2 over 3.0.1?
- Is your dynamically linked gdal library working if there's no [libproj](https://github.com/OSGeo/PROJ/) available on android? Didn't saw any libproj related stuff on the [diff](https://github.com/mavlink/qgroundcontrol/compare/master...Aeronavics:MapProviders_step2).
- Did you tested on android and it worked only with libgdal.so? If it worked it means that somehow libgdal.so is statically linked against libproj.a or something like this.

tilaktilak, 2019-10-08:
> @tilaktilak Nice work!
> I've managed to statically compile libproj (6.2.0) and gdal (3.0.1) on linux but the size of the binary is huge (93MB with most of the drivers disabled).

Same for me, here my options to compile gdal. I think I removed proj from the list as well.
[gdal_compile.txt](https://github.com/mavlink/qgroundcontrol/files/3704402/gdal_compile.txt)

 Let's have a call on skype(ID: cosmin_marc) and discuss more. I think we can work together, especially on the backend part.
> I have a few questions:
> 
>     * Why did you choose gdal 2.4.2 over 3.0.1?

For dependencies issues with gdal from ubuntu repos ( I think it was to keep the same OpenSSL version as the one in QGC), And also API and doc are not aligned with versions. We can probably pick another version though, but I've tried several ones and it tend to get bigger with years ^^.
 
>     * Is your dynamically linked gdal library working if there's no [libproj](https://github.com/OSGeo/PROJ/) available on android? Didn't saw any libproj related stuff on the [diff](https://github.com/mavlink/qgroundcontrol/compare/master...Aeronavics:MapProviders_step2).

Not sure actually, the app launch but did'nt have console to check if things works, but I am not sure I do need libproj for what i'm doing though. 

> 
>     * Did you tested on android and it worked only with libgdal.so? If it worked it means that somehow libgdal.so is statically linked against libproj.a or something like this.

Here my example code :
[geotiff2.txt](https://github.com/mavlink/qgroundcontrol/files/3704412/geotiff2.txt)
I compile it with : `g++ geotiff2.cpp -std=c++11 /pathtomylib/libgdal.a -ljson-c -lgeos_c -lpthread -ldl`

tilaktilak, 2019-10-09:
Got some good news : 
I confirm that I don't need libproj to extract elevation data from a geotiff.
I recompiled gdal (.so) with the minimal set of components (gdal_compile.txt in my previous post).   

my apk is **37Mb** and successfully open a geotiff and extract some data (the min and the max elevation.)

So compiling a minimal gdal + linking dynamically seems great option to minimize apk size.  

[ Another thing that might be not related : I can't open a QGCFileDialog from my Android phone. ]

tilaktilak, 2019-10-10:
The segfault was due to libproj not beeing present. Calls to Coordinate transformation functions ended up with segfault : https://gdal.org/tutorials/osr_api_tut.html?highlight=ogrcreatecoordinatetransformation#coordinate-transformation

tilaktilak, 2022-12-15:
Hi serbinsh,
Thanks for your message. 
This PR contained a rework of the terrain provider (following the first PR I did on map providers) + the integration of GDAL and LibProj used to load the DEMs, so pretty much all the backend to load a DEM and use it while mission planning.
Since then, I started a development services company to do this kind of work, I will have more capacity by the beginning of the year, I'll be happy to get in touch if you think you could be interested.
Otherwise, happy to assist you if you're willing to carry on and get this PR ready by yourself.

serbinsh, 2022-12-15:
@tilaktilak Thanks for the quick reply! Understandable and thanks for letting me know the state of this PR.  I would be happy to help in the new year if I can so consider me interested in the chance that we can talk next year about what it would take to complete this.  email: sserbin@bnl.gov

Ryanf55, 2024-04-12:
Sounds good. What would you consider a minimal PR suitable for merge? I propose the following starting with the build system:
* GDAL included as a submodule, or through find_package
* Headers get resolved
* Library can be linked to 
* Compiles on major platforms
* Includes a CMake Optoin to disable GDAL support

DonLakeFlyer, 2024-04-12:
It seems like I remember from way back when that GDAL wasn't supported on all QGC platforms.

Davidsastresas, 2024-04-13:
I took a look at it kind of recently, and I think they can be built for win, linux, and android but I am not sure how feasible it is for macos and ios. If it is not, should we go ahead anyway and just do a GDAL_DISABLED for mac and ios?

I would approach it as a submodule, as Ryan says, but maybe we can have the binaries directly on that submodule, similar to how we do with mavlink. GDAL version won't change for a long time for us probably, so I think it makes sense to not build it as part of QGC.

From there, ideally it would be great if we could have CI building the GDAL libraries on that submodule, so we have a reproducible way of building them.

dogmaphobic, 2024-04-13:
Isn't GDAL in QGIS, which is a Qt application that runs on macOS (in addition to Windows/Linux)?

Ryanf55, 2024-04-13:
> Isn't GDAL in QGIS, which is a Qt application that runs on macOS (in addition to Windows/Linux)?

[QGIS is available on Windows, macOS, Linux, Android and iOS.](https://www.qgis.org/en/site/forusers/download.html)

Yes, and I maintain a repo that is using it for MacOS. It [required special include paths for APPLE](https://github.com/ethz-asl/grid_map_geo/blob/02189d61cdc30c384e59a76ba1d044e002c39ba9/src/grid_map_geo.cpp#L47), otherwise was the same.
Should I just submit a PR to see what passes, or should I set up the toolchain to build all the platforms? 

The workflow I am using was standard CMake, so it should be platform-portable.

DonLakeFlyer, 2024-04-13:
My suggestion would be to wait until the full cmake conversion is done. And then submit a pull with cmake build. That way you don't need to double up on the build system work.

Davidsastresas, 2024-04-20:
Sounds good. How far are we from cmake conversion done? We will deprecate completely Qmake right?

DonLakeFlyer, 2024-04-23:
Remaining cmake steps:
* Switch over Mac builds to cmake in CI. Everything else is already there. I'm hoping I can finally finish this up in a few days time.
* Figure out new custom build cmake setup. This is going to take longer.
* Remove qmake. Not quite sure when we finally make it all the way here

Ryanf55, 2024-04-23:
> Remaining cmake steps:
>
> * Switch over Mac builds to cmake in CI. Everything else is already there. I'm hoping I can finally finish this up in a few days time.
> * Figure out new custom build cmake [...]

Awesome. Is there an issue or PR tracking these items? I can lend some time/expertise.

DonLakeFlyer, 2024-04-23:
> Is there an issue or PR tracking these items?

No, there isn't. I you are a Mac person I could use help with that. So far it's been making me crazy. But I don't have much cmake experience.

Ryanf55, 2024-04-23:
I can help with CMake, but I don't have a Mac.

DonLakeFlyer, 2024-04-23:
Ok. I'll push up a pull that I'm working on later today you can look at

Ryanf55, 2025-01-26:
CMake's working now. Looks like you all are changing quite a bit with fetching. How would you like this completed?
If you don't have a preference, I recommend using find_package, and requiring GDAL version 3.5 or higher, but less than 4.

gugpilot, 2025-06-14:
Hi, sorry, but do you have any news about importing a custom DEM? In Mission Planner, is it possible to put the GeoTIFF of custom elevation in the right directory? Something similar? Or ad geotiff import? What format is the ground control  elevation data? Can I put a geoserver source to download the custom elevation? Many thanks

Ryanf55, 2025-06-16:
> Hi, sorry, but do you have any news about importing a custom DEM? In Mission Planner, is it possible to put the GeoTIFF of custom elevation in the right directory? Something similar? Or ad geotiff [...]

Please don't use the QGC issues list for support on mission planner. They are maintained by different people. 
Custom terrain is not supported by QGC; this issues relating to it are still open.