mavlink/qgroundcontrol #7775: Layer shown while editing Mission/Fence/Rally

tilaktilak, 2019-09-05:
This issue is a follow-up to the discussion from #7765

When we edit the Fence or Rally point the Mission Layer is hidden. It's useful to show all layers not depending on what we are editing (Fence, Rally or Mission). Other layers that are not editable at the moment could be put in gray.

Need to be done in accordance with @DonLakeFlyer work on complex item visuals.

DonLakeFlyer, 2019-09-05:
So the problem here is working through the clutter of showing everything on the screen. Does it prevent you from getting to the fence or rally point item on the screen that you need to interact with. 

I think fence is likely easier since a fence would normally be outside the area of a mission. But even with that, getting to the polygon tool in the center may not be easy because it is covered up by other stuff.

Rally points could be much worse with a complex mission like a survey in combination with a survey for example.

To make it all work it may require dynamic manipulation of the z-order of elements such that the active editing layer elements are moved to the top of the z-order.

DonLakeFlyer, 2019-09-05:
There is also the issue of the "current Item" for mission items. I've changed the complex item code such that it show more minimal ui when the item is not current. That minimal ui should also be shown if the editing layer is not Mission. But requires extra work.

tilaktilak, 2019-09-05:
> So the problem here is working through the clutter of showing everything on the screen. Does it prevent you from getting to the fence or rally point item on the screen that you need to interact [...]

Yeah, I don't have something really specific in mind at this stage but the goal is to keep kind of a shadow of the other layers on the map but they are not editable.

tilaktilak, 2019-09-05:
> There is also the issue of the "current Item" for mission items. I've changed the complex item code such that it show more minimal ui when the item is not current. That minimal ui should also be shown if the editing layer is not Mission. But requires extra work.

I think it's something we have to keep in mind doing that but it's related to https://github.com/mavlink/qgroundcontrol/pull/7779 only right ?

DonLakeFlyer, 2019-09-06:
> I think it's something we have to keep in mind doing that but it's related to #7779 only right ?

No. It has to with Plan view. Look here: https://github.com/mavlink/qgroundcontrol/pull/7746

DonLakeFlyer, 2020-11-20:
Fence/Rally layer now shows the mission

closed 2020-11-20 by DonLakeFlyer