mavlink/qgroundcontrol #7776: Waypoint edition error _root is not defined

tilaktilak, 2019-09-05:
Discussed here : #7765

When I click to add a new Waypoint, this error shows up :
`qrc:/qml/QGroundControl/Controls/ToolStrip.qml:87: ReferenceError: _root is not defined`

Might be linked to this commit : https://github.com/mavlink/qgroundcontrol/commit/6940e70c1e91bfdc834bdbfce8c19231a94db330

Looks like _root is not defined for items of the ToolStrip without dropPanelComponent (like WP)  

@dogmaphobic if you can give me some inputs on that ... :-)

tilaktilak, 2019-09-06:
Not sure but that could do the trick : 
https://github.com/mavlink/qgroundcontrol/commit/74527fd0360d498ebce3efa72c1cf9820f0a4a65

DonLakeFlyer, 2019-09-06:
As far as I can tell this:
```
                    if(_root && buttonTemplate)
```
 Doesn't really make sense. ```_root``` is an ```id:```. You can't do an if on that and it will always be there.

DonLakeFlyer, 2019-09-06:
Wait ```buttonTemplate``` is an id as well?

DonLakeFlyer, 2019-09-06:
@dogmaphobic Is going to have to comment. Not sure what he is trying to accomplish with this if testing.

tilaktilak, 2019-10-14:
@dogmaphobic have you had a chance to take a look at this ?
https://github.com/mavlink/qgroundcontrol/blob/master/src/QmlControls/ToolStrip.qml#L87

closed 2026-10-04 by github-actions[bot]