mavlink/qgroundcontrol #7777: Spline or Land can result in WP at (0,0)

tilaktilak, 2019-09-05:
Discussed here : #7765 

If we change the type of the Takeoff item (just after the Mission Start) to spline or land (Maybe some other type of WP), it create a WP at (0,0)
[image]

DonLakeFlyer, 2019-09-05:
Ah, I wasn't reading this correctly initially. That's a nasty bug which has had various incarnations. Though I had found/fixed them all. Guess not. It's quite involved and relates to converting a mission item from on type to another and correctly applying the set of defaults again it. I'll probably have to deal with that part.

I think there may be another bug here as well. Does ArduPlane support a Takeoff item which includes lat/lon. I think QGC has that as no lat/lon.

tilaktilak, 2019-09-05:
Not sure if I see what you mean here, but for reference : http://ardupilot.org/plane/docs/takeoff-mode.html
Also we are using ArduCopter and I was testing with Copter simu. But to reproduce this bug you don't need to connect any device.

DonLakeFlyer, 2019-09-05:
Here is an example of a simple fixed wing plan for PX4:
[image]

The vehicle takes off from the ground and heads towards the takeoff item location and height.

In ArduPilot the takeoff item has no lat/lon for a fixed wing. According to the link you have above I think that is just the way it works. So QGC is correct. But it also means that the flight path that QGC draws for this case will be incorrect. It will go from planned home to first waypoint. Not sure how other gcs like MP handle that case, or if they do.

PR #7787 (2019-09-07): Correctly handle command changes from no-coordinate item to coordinate
Fixes #7777
closed 2019-09-07 by DonLakeFlyer