mavlink/qgroundcontrol #7780: Flight Planner unable to save fences unless mission is also defined

rmackay9, 2019-09-06:
On the flight planner screen the "Plan" panel has a "File" selector button which opens up a window to "Save" the mission and fences to a file.  It seems the "Save" button is greyed out unless a mission is defined.  This means it isn't possible to save a file that only has fences defined.

DonLakeFlyer, 2019-09-06:
@rmackay9 Latest master/daily build right? I did this:
* Booted QGC without a vehicle connected
* Plan View
* Added a fence polygon
* Click File, see that "Save As" is enabled.
* Do a "Save As".
* Click File, now both "Save" and "Save As" is enabled.

Also tried booting with a vehicle connection with no Plan on it. Same thing.

Maybe some detailed sequence problem again.

DonLakeFlyer, 2019-09-06:
Trying some different ways to make it happen...

DonLakeFlyer, 2019-09-06:
@rmackay9 Note the sequence of Save As -> Save. Was "Save As" disabled as well?

DonLakeFlyer, 2019-09-26:
@rmackay9 Still happening?

DonLakeFlyer, 2019-12-11:
@rmackay9 Any updates?

rmackay9, 2019-12-12:
@DonLakeFlyer, ah, sorry, I didn't see the messages, I'll re-test.

DonLakeFlyer, 2019-12-24:
I've tested and it looks fine. Going close unless we get some concrete repro steps.

closed 2019-12-24 by DonLakeFlyer