mavlink/qgroundcontrol #7784: Set RC to Param not working in QGC v3.5

Molecuman, 2019-09-07:
I have this issue: Selecting "Advanced settings" -> "Set RC to Param..." in Parameter Editor is not working in QGC 3.5 with PX4 v1.9 on Pixhawk, tested on Windows 10. 

[image]

DonLakeFlyer, 2019-09-07:
It looks like you are using a Daily build not a Stable build. Set RC to Param is currently not supported in Daily builds. This is tracked here: https://github.com/mavlink/qgroundcontrol/issues/7770. Stable 3.5.4 build has been verified to work.

Are you using a Daily build?

Molecuman, 2019-09-07:
Hi. Thank you! I downloaded again the QGC and now it works fine.

closed 2019-09-07 by DonLakeFlyer