mavlink/qgroundcontrol #7791: Feature Request: add "current mission item" (or current mission index) to Value Widget list

Antiheavy, 2019-09-08:
Would be useful to see the active mission item/index in the values widget for certain situations.  Especially with the newer simplified Fly screen visuals, sometimes it is not obvious what the active mission item is.  Having this option in the Values Widget gives the advanced user an option to always see what is active, even if off screen or hidden or whatever.

[image]

DonLakeFlyer, 2019-09-09:
I would think that with the new visuals, the current mission item is way more obvious than it used to be. The only case would be if it is off screen. In general I'm moving away from showing mission item numbers anywhere. I just don't see who they add any value.

Other than offscreen, I don't quite get why current mission item is unclear?

Antiheavy, 2019-09-09:
In addition to off-screen, there are times when the current mission item is not a waypoint and is sucked up/hidden.  In particular before Takeoff.  When troubleshooting an issue it can be valuable to know if the mission is correctly indexed and starting at 1.  this request isn't for an always shown value, but one that would be turned on for troubleshooting or in special situations.

DonLakeFlyer, 2019-09-09:
> there are times when the current mission item is not a waypoint and is sucked up/hidden.

Hmm, I was think that all of these cases would only last for like a few milliseconds after which the next waypoint would go current. That may not be the case. Need to think about the other item types. May need to do something special here.

> When troubleshooting an issue it can be valuable to know if the mission is correctly indexed and starting at 1. this request isn't for an always shown value, but one that would be turned on for troubleshooting or in special situations.

That's all good.

Antiheavy, 2020-03-14:
Had a user request for this feature today.  Related to no longer seeing mission item numbers on the map.  

loosely related to: https://github.com/mavlink/qgroundcontrol/issues/8478

Antiheavy, 2020-03-29:
upstream PR merged.  thanks!  https://github.com/mavlink/qgroundcontrol/pull/8577#event-3175996117

closed 2020-03-29 by Antiheavy