mavlink/qgroundcontrol #7792: new mission line bisector not working right with Landing Pattern (and other complex patterns?)

Antiheavy, 2019-09-08:
If the Landing Pattern is active, and I click on the line bisector running to the Loiter, a line is created from the Land WP to the bisector.  I'm pretty sure this is not intended.
[image]

DonLakeFlyer, 2019-09-09:
Are you using a master from last night. I fixed a bunch of stuff around this yesterday. I tried the bisect while Landing pattern selected and it worked fine for me.

Antiheavy, 2019-09-09:
Still happening for me on today's daily (different laptop this time):
[image]

DonLakeFlyer, 2019-09-09:
Ah, sorry. It's the Survey item which is making this crap out. I thought it was just the Landing pattern. I can reproduce.

closed 2019-09-10 by DonLakeFlyer

Antiheavy, 2019-09-10:
yep.  fixed.  thanks.

PR #7803 (2019-09-10): Plan: Fix SplitSegment mission item insertion
Fixes #7792