mavlink/qgroundcontrol #7796: Label Takeoff waypoint (or indicate as special) in new Fly view visuals

Antiheavy, 2019-09-09:
I think the Takeoff waypoint needs something special indicating it as Takeoff in the new Fly screen visuals.  When using auto takeoff, it is important for the user to visually confirm the "Takeoff" waypoint is there and in the correct location they plan to takeoff to before they "Start Mission".  Also, it just looks sort weird at the moment:
[image]

Here is the Plan view for reference: 
[image]

DonLakeFlyer, 2019-09-09:
Yeah, I'm working all of this still. There are still multiple problems with it.

DonLakeFlyer, 2019-09-09:
I think a better way to do this is to only hide the generic waypoing indicators. And show the more "special" items with an indicator.

closed 2019-09-09 by DonLakeFlyer

Antiheavy, 2019-09-10:
yep, better.  thanks.
[image]

PR #7797 (2019-09-09): Only hide generic waypoint indicators
Fixes #7796

[image]