mavlink/qgroundcontrol #7799: ArduPilot: Takeoff must be first mission command

tilaktilak, 2019-09-09:
Following #7773 and #6776 

There is no TakeOff added if we create a mission with a complex pattern.
Need to insure a takeoff is added at the beginning of every ArduPilot mission.

edwinhayes, 2019-09-09:
Semi-related: has there been a change so that now the takeoff waypoint doesn't have a little circle with a "T" colocated with the preceding waypoint (which was usually planned home)?

tilaktilak, 2019-09-09:
Yes I think it has been discussed here : https://github.com/mavlink/qgroundcontrol/issues/7796
and fixed here : https://github.com/mavlink/qgroundcontrol/pull/7797

edwinhayes, 2019-09-09:
That discussion seems specific to the fly screen, whereas I'm talking about the plan screen?

DonLakeFlyer, 2019-09-09:
Yes the Plan screen visuals with respect to Takeoff items without a lat/lon is out of whack right now.

PR #7804 (2019-09-16): Add Takeoff SimpleItem if first item is complex
Fixes #7799

As it's done for SimpleItem insertion, add a SimpleItem with TakeOff command before the ComplexItem
closed 2019-09-16 by DonLakeFlyer