mavlink/qgroundcontrol #7800: Inconsistent behaviour for inserting waypoints.

edwinhayes, 2019-09-09:
When inserting waypoints via the hamburger menu, the location of inserted waypoints seems inconsistent; sometimes the new waypoint seems to be at a totally arbitrary location, other times it's colocated with the existing waypoint.

Also, there is no way of adding a new waypoint at the end of a mission other than by clicking on the map, since the hamburger menu's insert option only inserts a waypoint before the current item.  It would be nice to have "Insert before" and "Insert after" options.  For instance, this allows adding a landing command to the end of an existing mission without actually changing the last waypoint.

edwinhayes, 2019-09-09:
@tilaktilak

DonLakeFlyer, 2019-09-09:
> When inserting waypoints via the hamburger menu, the location of inserted waypoints seems inconsistent; sometimes the new waypoint seems to be at a totally arbitrary location, other times it's colocated with the existing waypoint.

The new waypiont should always end up in the center of the entire map. Is that not what you are seeing?

DonLakeFlyer, 2019-09-09:
> Also, there is no way of adding a new waypoint at the end of a mission other than by clicking on the map, since the hamburger menu's insert option only inserts a waypoint before the current item.

I looked at that a while back, but instead opted for keeping things simple. In reality the fact that Fixed Wing Landing is in there is a bug since that should always be at the end. With the new split segment support the hamburger will get even less use. So I don't see a lot of value in making it do more things.

edwinhayes, 2019-09-09:
> The new waypiont should always end up in the center of the entire map. Is that not what you are seeing?

Oh, yes, I see (it looks offset, because the map extends behind the list of mission items on the RHS).  Still, that's seems totally arbitrary and not very practical.  What was the reason for this?  Colocated with the point it was inserted before seems more practical, because if you're using the hamburger menu, it's probably to insert a mission item which you couldn't just have inserted by clicking on the map (probably because you want it colocated).

> So I don't see a lot of value in making it do more things.

Being able to add colocated mission items is a pretty common use case for Ardupilot users.  Not being able to add a colocated waypoint at the end means that you can't easily make a mission which flies to a specific spot, and _then_ lands at that same place.

tilaktilak, 2019-09-09:
Maybe this 'colocated mission items' could all mission item that does not specifies coordinates ? and we associated them to the position the drone will be when this will occur ?

DonLakeFlyer, 2019-09-09:
> Still, that's seems totally arbitrary and not very practical.

It's arbitrary since that was the base information I had available in the previous code. With what I have I think I can make it work just like the split segment works.

If CoLocation is needed then I think that should be supported using something else. The current ui isn't really meant to support that.

DonLakeFlyer, 2019-09-09:
A better way to discuss these sort of problems is to use examples which were created from Mission Planner, attach the mission plan to an Issue and then explain why this is a royal PITA to create the same thing in QGC. Then I have a better idea of what you want.

tilaktilak, 2019-09-09:
Not sure if it's what you mean Edwin but a good example could be with camera control : 

If I want to add a "Camera Trigger Distance" on the last WP I can either :
 - Change the type of the last item in the list ( but the WP will disappear ) 
-  insert a WP via the Hamburger menu that will temporarily appear on the center of the map then change its type, then it will disappear.

The expected behavior should probably be a tag on the previous WP that appears when we add a missionItem that does not specifies coordinates (Same as the Takeoff one) ?

@edwinhayes  can you confirm this is what you referring ?

edwinhayes, 2019-09-09:
1. In general, I think what you are describing would be sensible: that mission items that weren't waypoints appeared on the map has an annotation/tag to the previous waypoint (I'd suggest a circle which appeared tangent to the waypoint in a grey callout same as the text "Planned Home" appears in - it could be hidden when the waypoint wasn't selected).

2. This does raise the issue though, that QGC can present the same mission in two ways; when you create a waypoint which is actually a composite of waypoint plus gimbal plus cam trigger, that presents differently to when you manually make a mission which encompasses these items.

3. There are cases for colocated regular waypoints though: fly to a particular spot, then climb vertically to a certain distance, then wait a while, then climb further, then wait a while, then climb further, etc.  That's just one example which I know is a core use case of one of our customers, is currently annoying to perform in QGC, and would trivially be made really easy if simply there was an option to add a waypoint to the end of the list, and if the default position was colocated with the previous waypoint.

DonLakeFlyer, 2019-09-09:
> 2\. This does raise the issue though, that QGC can present the same mission in two ways; when you create a waypoint which is actually a composite of waypoint plus gimbal plus cam trigger, that presents differently to when you manually make a mission which encompasses these items.

Only at first creation time. If you reload the Plan those extra items get slurped up into the previous waypoint.

DonLakeFlyer, 2019-09-09:
> If I want to add a "Camera Trigger Distance" on the last WP

You do it this way:
[image]

edwinhayes, 2019-09-09:
> Only at first creation time. If you reload the Plan from the vehicle or off disk those extra items get slurped up into the previous waypoint.

In that case, a possible improvement might be to expand the behaviour to happen automatically, so that if you try to add one of the aggregate type items, it just gets 'sucked' into the previous waypoint?

That would at least address the issue of having the same mission rendered in two ways.  Though you'd still have to cater to other mission items without positions, because I suspect it's probably not feasible to support every command type which Ardupilot supports with the nice 'wizard' UI.

DonLakeFlyer, 2019-09-09:
> Though you'd still have to cater to other mission items without positions, because I suspect it's probably not feasible to support every command type which Ardupilot supports with the nice 'wizard' UI.

The way this works doesn't really have anything to do supporting the set of mission items which have no position associated with them. This feature was meant to make two specific use caes easier:
* Working with a cameras
* Flying to a survey area fast and then slowing down ocne you get to the survey. Hence the corresponding item in Mission Start.

DonLakeFlyer, 2019-09-09:
It's all based on the fact that most people don't understand the concept of how missions actually work in that they are individual commands which run insequence. Nor do they want to understand that in most cases. Just just want to fly somewhere and take a picture.

closed 2026-10-04 by github-actions[bot]