mavlink/qgroundcontrol #7801: OpenSSL lib file is missed for Windows 10 Build

jeyong, 2019-09-09:
- Operating System: [Windows 10]
- QGC Version: [master branch]

I met compile error.

master branch build error :
Build error : "LNK1104: cannot open file 'crypto.lib' 

I guess that crypto.lib should be qgroundcontrol/libs/OpenSSL/Windows/x86/lib/ for Windows 10 build.

tilaktilak, 2019-09-09:
Could probably be related to #7798

DonLakeFlyer, 2019-09-22:
@jeyong Are you compiling without gstreamer?

DonLakeFlyer, 2019-09-22:
Yup, the build randomly picks up crypto.lib from the gstreamer install which is incorrect. I think the same thing is happening on Ubuntu 18.04 where I can get it to compile either.

DonLakeFlyer, 2019-09-22:
I've been having problems with it on Android as well. Maybe the same incorrect gstreamer connection there to. Don't know.

jeyong, 2019-09-22:
@DonLakeFlyer I also try compiling with gstreamer on Windows 10. I got same error. But it compile on Ubuntu 18.04, it is successful.

jeyong, 2019-09-22:
@DonLakeFlyer I tried to install gstreamer-1.0-devel-mingw-x86-1.16.0.msi and gstreamer-1.0-devel-msvc-x86_64-1.16.0.msi. Any recommended version or configuration?

DonLakeFlyer, 2019-09-23:
Follow the instructions here: https://github.com/mavlink/qgroundcontrol/blob/master/src/VideoStreaming/README.md

jeyong, 2019-09-24:
@DonLakeFlyer Thanks. Now it works.

closed 2019-09-24 by jeyong