mavlink/qgroundcontrol #7802: Plan: Make Insert Waypoints from hamburger work like split segment

DonLakeFlyer, 2019-09-09:
Now that Split Segment support is in there we can make Insert Waypoint smarter as opposed to the dumb way it works now which is to position the new waypoint in the center of the map.

Related to #7800

edwinhayes, 2019-09-09:
Why is having the behaviour of the hamburger menu the same as clicking the split segment desirable?

If you just wanted to add a waypoint splitting a segment, there is already a superior UI mechanism for that: the button on the map.

But, if you specifically want the new waypoint to be colocated, this change would mean it's no longer possible at all.

DonLakeFlyer, 2019-09-09:
> If you just wanted to add a waypoint splitting a segment, there is already a superior UI mechanism for that: the button on the map.

Good point. Let me think about the co-location thing then.

Antiheavy, 2019-09-10:
>  if you specifically want the new waypoint to be colocated,

I'm curious what you mean by this.  When I "insert" a waypoint it always just plops down in the center of the screen.  How do you colocate mission items? any what use cases?  I'm honestly asking because I'm curious.

DonLakeFlyer, 2019-09-10:
Say you want to fly to a location at specific altitude, then go up 10 meters at same location, then up 10 meters again at same location. All the waypionts are on top of each other at same lat/lon.

DonLakeFlyer, 2019-12-11:
@edwinhayes So this is all different now on latest daily. Is there still a problem with creating a col-located item?

edwinhayes, 2019-12-11:
@DonLakeFlyer Hi Don; since I've stepped back from my role at Aeronavics, the best person to follow this up with would be @tilaktilak.

tilaktilak, 2019-12-12:
I have tested master this morning.
I did not find any method to create collocated WP unless editing the position manually. 

Also, the Takeoff point is offset from the first WP (or Home position) when I create a Blank misison (Ardupilot Mission) : 
[image]

DonLakeFlyer, 2019-12-12:
> Also, the Takeoff point is offset from the first WP (or Home position) when I create a Blank misison

That is correct behavior. If the first thing you add is a waypoint as opposed to a takeoff item then the mission is assumed to start from the air. In which cause the launch point would differ from  the first waypoint.

closed 2019-12-12 by DonLakeFlyer