mavlink/qgroundcontrol #7805: 3.5.4 Not connecting to PX4 1.10.0.beta in Ubuntu 18.04

ealdaz, 2019-09-10:
## Expected Behavior
QGround control should connect to Pixhawk when connecting USB (no other source of power)
## Current Behavior
No progress bar visible, left menu not updated, but there seems to be some sort of connection as get error messages

## System Information
Ubuntu 18.04
Pixhawk 2.1 (Cube)
PX4 v1.10.0.beta

Just reporting in case it's useful. This happens on two different Ubuntu 18.04 computers.
Version 3.5.3 works fine.

DonLakeFlyer, 2019-09-10:
* Can you turn on LinkManagerLog: https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html
* Then boot QGC with the board connected
* Go to Settings/Console, save the output and attach here

ealdaz, 2019-09-11:
Hello Don

We’ve changed firmware a couple of times since my initial test, reinstalled  V1.10.0beta and it works fine with v3.5.4, this is good news.

I suspect in our initial tests we might have been in the latest dev master and not in V1.10.0beta like I thought, the confusion likely to be caused by the fact that Firmware Version(in Summary Tabs) reports the latest tag even if the firmware is actually a newer commit. i.e the commit we used in firmware is more recent than the V1.10.0beta release, but perhaps because it doesn’t have a tag is the latest existing tag that is used instead.

I don’t know if the commit number is reported by the firmware, but if it is it would be great to add it to the summary page.

THanks, and apologies for the false alarm

Eduardo

DonLakeFlyer, 2019-09-11:
Ok, great thanks for letting me know

closed 2019-09-11 by DonLakeFlyer