mavlink/qgroundcontrol #7812: [Feature Request] Have QGC announce VTOL transition completion

dlwalter, 2019-09-13:
It is helpful for our operators to have QGC announce when VTOL aircraft have reached "Fixed-Wing Mode" or "Rotary-Wing Mode" at the end of a front or back transition.

We currently implement this using "mavlink_log_critical()" function calls in our vtol_att_control class within the update_vtol_state state machine.

Is there a way to have QGC do this on its own?

DonLakeFlyer, 2019-09-13:
I'd rather keep QGC dumb if possible. This keeps thing simpler when the firmware side changes and/or mavlink messaging. At this point QGC announces next to nothing on it's own other than flight mode and arm states. Why is it bad to have the vehicle say this? There are tons of other cases where the vehicle announces things.

DonLakeFlyer, 2019-09-26:
I"m closing unless there is better reasoning as to why the firmware shouldn't handle this.

closed 2019-09-26 by DonLakeFlyer