mavlink/qgroundcontrol #7820: Elevation data not accessible offline

seangrant82, 2019-09-19:
This system is for posting bugs or feature requests ONLY. For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

Problem:
When exporting map tiles and elevation data into a qgctiledb, when try to import that qgctiledb back into QGC, the elevation does not import. This prevents any sort of terrain following in an offline mode. 
QGC version: 3.5.4 (latest stable release from website on windows and linux tested)
Steps to reproduce:

1. Open QGC and ensure you are connected to the internet
2. Click the main QGC icon in the top right to access the settings
3. Click Offline Maps
4. Click Add New Set
5. Zoom into a particular area and and set your min and max zoom, name and Map Type. Ensure the box to Fetch elevation data is checked.
6. Click Download
7. Once data is downloaded and indicator is green, click Export
8. Select the tile set and elevation you just downloaded and click Export
9. Choose a name and location for your file
10. Once the export is complete click close on the export dialog and cancel in the export menu
11. DIsconnect from the internet
12. Click the arrows on the right of your tileset and elevation you just downloaded and click delete
13. Once they are delete Click import and choose the qgctiledb you exported earlier.

You will see only the tile are imported not the elevation.

DonLakeFlyer, 2019-09-19:
@tilaktilak Have you gotten close to the import/export code? Wondering if you could take a look at this?

tilaktilak, 2019-09-19:
Yep, I'll be glad to have a look !

seangrant82, 2019-09-19:
@tilaktilak thx! let me know if yo need me to test anything or need any clarification

tilaktilak, 2019-09-19:
I can't reproduce this easily, can you share the qgctiledb file with us ?
Can you confirm that in this particular area and with internet connected, when you create a mission the Elevation graph on the bottom-left corner is updated with terrain data ? 
[image]

seangrant82, 2019-09-19:
- Here is one I made over the grand canyon. [gc2.zip](https://github.com/mavlink/qgroundcontrol/files/3633410/gc2.zip)
- With alt above terrain and internt connected
[image]

- With altitude bot above terrain and internet connected 
[image]

-I don't receive and error message. When I import the qgctiledb only the map tiles get imported. The elevation does not. Here is screenshot of me importing the db above 
[image]

tilaktilak, 2019-09-19:
```
sqlite> SELECT * FROM TileSets
   ...> ;
1|gc|Bing Hybrid Map|36.1026981191769|-112.139439165652|36.0941894090616|-112.12205650805|16|17|666|47|0|1568822932
2|gc Elevation|Airmap Elevation Data|36.1026981191769|-112.139439165652|36.0941894090616|-112.12205650805|1|1|8001|4|0|1568822932
sqlite> SELECT * FROM TilesDownload;
sqlite> SELECT * FROM Tiles;
1|06660001235300025711016|jpg|����|23503|666|1568822932
2|06660001235300025712016|jpg|����|19757|666|1568822932
3|06660001235300025713016|jpg|����|24590|666|1568822932
4|06660001235400025711016|jpg|����|25839|666|1568822932
5|06660001235400025712016|jpg|����|21865|666|1568822932
6|06660001235400025713016|jpg|����|17510|666|1568822932
...
42|06660002471200051426017|jpg|����|17546|666|1568822932
43|06660002471200051425017|jpg|����|24902|666|1568822932
44|06660002471300051423017|jpg|����|19036|666|1568822932
45|06660002471300051426017|jpg|����|14813|666|1568822932
46|06660002471300051424017|jpg|����|20097|666|1568822932
47|06660002471300051425017|jpg|����|21830|666|1568822932
```

From what can I see in your db file, there is no elevation tiles, and nothing in the TilesDownload table (which is the "To download" list). 
Whereas the TilesSet exists and should contains 4 tiles.

seangrant82, 2019-09-19:
[example.zip](https://github.com/mavlink/qgroundcontrol/files/3633546/example.zip)
Can you try this one?
It appears the export isnt grabbing the tiles when it should. I looked at this and there are two tilesets

tilaktilak, 2019-09-19:
I confirm this one has 2 elevation tiles and 14 Google Street Map ones

seangrant82, 2019-09-20:
I am only selecting the ones that I downloaded. Yeah, I found if the button is disabled, i check another tileset and uncheck it and that appears to enable it

closed 2026-10-04 by github-actions[bot]