mavlink/qgroundcontrol #7825: Can't see "Camera" section

tuloski, 2019-09-23:
Probably it's me that I forgot to enable something but I don’t see the camera (https://docs.qgroundcontrol.com/en/SetupView/Camera.html) section in Qgroundcontrol. I see a reference to “PX4 Pro”. What is px4 pro? And is it normal that the camera section is not displayed?
Also all the params related to camera (TRIG_XXX) are not present.
I'm using latest (1.9.2) firmware for multicopter.
Qgroundcontrol is latest 3.5.4.

DonLakeFlyer, 2019-09-23:
Camera support will only show up if the TRIG_MODE parameter is visible. There is something on the firmware side which turns this on. You'll need to ask the firmware folks what it is.

closed 2019-09-23 by DonLakeFlyer

hamishwillee, 2019-09-24:
FYI PX4 Pro is the same as PX4.
There are some docs on what it means when a parameter is missing from QGC here: http://docs.px4.io/master/en/advanced_config/parameters.html#missing

Essentially either they are not in the firmware OR they are conditional on another parameter being enabled.

tuloski, 2019-09-24:
I have latest firmware and in the full parameters list it's not written it's conditional to another parameter.
Should I ask this on PX4/Firmware?

hamishwillee, 2019-09-25:
@tuloski Don't take it personally, I hate everyone :-)

So what exactly was the problem in the end/what note were you thinking? 

If you were using FMUv2 then this problem is fixed by the [addition of this section](http://docs.px4.io/master/en/advanced_config/parameters.html#missing). I don't want to add specific docs in  camera as so much is missing from fmuv2. 

Or was the problem that TRIG_MODE is not sent on boot? In this case then "perhaps" - did you fix it using the suggestion to set `param set TRIG_MODE 1`?

@DonLakeFlyer I am confused by this issue - aren't all parameters sent on boot, even those that default to zero? Or are you saying that this parameter is sent, but for some reason QGC treats it as zero (I can't even reproduce this, as simulator always shows camera).
Basically I'm trying to work out if this is a bug in PX4, or a feature (if a bug I will post a defect report).

tuloski, 2019-09-25:
@hamishwillee basically pixhawk 2 (and other autopilots) don't have enough memory to support all modules so camera module is disabled. This info (which modules are supported by which hardware) is not written anywhere. To me it's enough to say that camera module might not be supported by some hardware.

DonLakeFlyer, 2019-09-25:
> I am confused by this issue - aren't all parameters sent on boot, even those that default to zero?

It depends on your definition of "all". If your definition of "all" is all the possibile parameters that exist, then the answer is no. If you definition of "all" is what the firmware thinks are the ones that are currently in use for the way the firmware is configured then the answer is yes.

> Or are you saying that this parameter is sent, but for some reason QGC treats it as zero

No. The TRIG_MODE parameter in this case if not being sent by the firmware to QGC. I vaguely remember the fact that some other parameter controls whether it shows up or not. Also keep in mind that if parameter A is hidden behind the setting or parameter B it will require a reboot of firmware to get parameter A to be sent. Theoretically parameter B should then have "reboot required" meta data set on it in which case QGC will tell you that. But it may be missing, which would be a firmware bug against meta data.

DonLakeFlyer, 2019-09-25:
Oh, and there is also a concept on the firmware side of all parameters being "optional". I think it still works this way. Which means that if the firmware code doesn't touch a parameter at boot it won't be sent either.

DonLakeFlyer, 2019-09-25:
Sorry missed the thing about fmuv2 not supporting camera. Anyway hopfully useful information for you @hamishwillee.

tuloski, 2019-09-25:
@DonLakeFlyer about the metadata, QGC not always shows the correct info :), but this is a different topic.

hamishwillee, 2019-09-26:
> This info (which modules are supported by which hardware) is not written anywhere. 

@tuloski The FMU2 limitation is mentioned in heaps of places. I didn't want to mention the camera issues specifically because in reality this could affect any module. However since the camera is the one that has come up I have added notes as requested:
- QGC: https://docs.qgroundcontrol.com/en/SetupView/Camera.html#px4-camera-setup
- PX4: http://docs.px4.io/master/en/peripherals/camera.html

@DonLakeFlyer Thanks for your help. The background is useful (and has to some extent been captued in the PX4 "finding a parameter" topic: http://docs.px4.io/master/en/advanced_config/parameters.html#missing