mavlink/qgroundcontrol #7833: No Killswitch for Joystick Control(Feature Request)

jcharette570, 2019-09-27:
Could not locate a Feature Request form, if this is misplaced please let me know and I'll correct it.
## Expected Behavior
There should be a way to map a Killswitch to a joystick input button.  Disarming is passed through checks before it is allowed to stop the motors/actuators.  Killswitch simply stops all pwm outputs.

## Current Behavior
No killswitch available, with excess vibration the autopilot software(PX4 and maybe others) sets the state to "Not Landed" and will disallow disarming.  This is a safety concern.  If the vehicle has a collision or something breaks loose it can induce vibrations at which point the only way to stop the motors is to pull the battery.

## Steps to Reproduce:
Using PX4, QGC, and a Joystick
1. Arm vehicle (Without Props[Safety First])
2. Shake vehicle to simulate vibrations.
3. Attempt to disarm vehicle while shaking.
4. You should see a warning; "Vehicle not landed! Disarming Denied" with the motors still spinning.

## System Information
- PX4(Autopilot)  Ubuntu 16.04
- QGC Version: [3.5.4]
- QGC build: stable
- Flight Controller: mRo PixRacer
- Autopilot PX4 v1.9 & Daily
## Detailed Description
It shouldn't be too difficult to add the killswitch function to the drop-down list of joystick trigger options, as opposed to only being available via RC.  I had an airframe with vibration issues and after I armed the vehicle to attempt a first flight test, I was unable to disarm, and was forced to manually pull the power.  I was in a controlled environment so it was not devastating, but I can imagine situations where this could be a major safety concern.  I posted to PX4 forums about it  and was forwarded here as it is more a QGC request than PX4.

DonLakeFlyer, 2019-09-27:
> It shouldn't be too difficult to add the killswitch function to the drop-down list of joystick trigger options, as opposed to only being available via RC.

True. But my main concern here is preventing accidental triggering. To me that means it can't just be a single button press. And accidential press could be catastrophic.

DonLakeFlyer, 2019-09-27:
Also keep in mind that you can Emergency Stop using the QGC ui. Click Disarm on the toolbar.

jcharette570, 2019-09-27:
I do see your point in the safety aspect of accidental killswitching.  I also wasn't aware that the disarm in QGC was different than a joystick disarm.  You're saying the gui disarm wouldn't be blocked by the firmware during a situation where it is believed that the vehicle has not landed? This would lul my concerns.  I switched to a modified version of PX4 for the interim, that disables the feature that blocks disarming during flight.  Thank you for the prompt response.
Jake

DonLakeFlyer, 2019-09-27:
If the vehicle is armed but reported as in the air: Clicking Disarm will change to an Emergency Stop slider.

jcharette570, 2019-09-30:
Ok, Awesome, I wasn't aware of that, it should handle all my concerns.  Thanks for the info!

closed 2019-09-30 by jcharette570