mavlink/qgroundcontrol #7835: Current Daily: after placing a survey, Waypoint tool becomes active

Antiheavy, 2019-09-28:
In the current daily build, after placing a Survey Pattern, the Waypoint button becomes active.  This means if the user clicks on the map a waypoint appears and kicks them out of the survey settings.  This is too easy to accidentally do and if the user doesn't notice the extra waypoint added this could lead to confusion and problems.
[image]

Antiheavy, 2019-10-11:
This also happens in the new Plan Wizard and is very confusing for the user as disconnected waypoints start getting added after the Landing Pattern and it's unclear to the user why just clicking on the map started adding these.
[image]

DonLakeFlyer, 2019-12-11:
Should be fixed now

closed 2019-12-11 by DonLakeFlyer