mavlink/qgroundcontrol #7842: rework video streaming code

jaxxzer, 2019-10-01:
The video receiver code is a mess of spaghetti, and video streaming has suffered a long running blight of bugs and spotty support. It needs some tender love and care.

I think that we should define exactly what we want to achieve, make a plan and execute it. @tcanabrava will be available to help us with the api design and execution.

I propose this list as a general starting point to aim for in terms of functionality. Some of these features are already (dubiously) implemented:
- Support formats: H264/265, MJPEG, YUYV (raw)
- Support RTSP and UDP
- Support any resolution
- Support at least 4 concurrent streams, display and recording
- Solid support for video streaming on all of our platforms

This is open for discussion and input. I'd like to establish what the community needs and what will be accepted for merge. We have some preliminary UML work to demonstrate how this could be accomplished, and we will update this as it develops.

DonLakeFlyer, 2019-10-01:
"and crashes"

@dogmaphobic Needs to be directly involved in this

MatejFranceskin, 2019-11-12:
I am also interested in this. Would it be possible that you share perliminary UML work?

tcanabrava, 2019-11-12:
There’s an open wip PR that you can check

jaxxzer, 2019-11-12:
The wip pull request is [here](https://github.com/mavlink/qgroundcontrol/pull/8017).

@tcanabrava can you post your UML here?

jaxxzer, 2019-11-25:
ping @tcanabrava

tcanabrava, 2019-11-26:
Right now I have two branches, one with a simplified camera handling (handles *one* camera) and another with a complex camera handling (handles 40+ cameras).

[image]

The simplified camera handling has the following features:
- play video
- pause video

Rationale:
- The old QtGstreamer code is removed in favor of the new QmlGlSink, this alone removes around 6k lines of code, simplifying things.
- The new QmlGlSink works in OpenGL so we have a *much* nicer CPU use as there's no need to copy bitmap buffers all around
- The pipeline is created via gst_parse, significantly simplifying pipeline code (and if we know the pipeline we can add only the plugins that we use for the binary, no need to add the full gstreamer)
- Play / Stop happens in OpenGL render thread, this means that we can't call ->play() ->stop() directly as we used to do, communication should be done by signals / slots (that also helps cleaning out code and detangling - for instance, in master vehicle calls camera (but shouldn't)).

[image]

The Multiple Camera approach is build on top of the single camera approach, but adds a few other things:

- A model that controls the amount of cameras that are being currently displayed
- Settings *per* camera
- Control of the video *per* video item.
- (tested in a i5 laptop and we could easily handle 40 streams)

tcanabrava, 2019-11-27:
[image]

tcanabrava, 2019-11-27:
This draft uml does not represents correctly the current implementation as the current implementation is much more complex, but it gives us a more understandable approach of the current architecture of video in qgc.

closed 2026-10-04 by github-actions[bot]