mavlink/qgroundcontrol #7853: Rover: throttle values are sent as -100 to +100 and QGC rolls percentage over for negs

Hwurzburg, 2019-10-02:
negative throttle percentages (reverse) are sent by APMRover2 to hud display as a signed integer in an unsigned type....AP OSD displays them correctly, but QGC rolls negatives roll over to 65000 range...

DonLakeFlyer, 2019-10-02:
@hamishwillee This is caused by the fact the VFR_HUD.throttle is a uint16_t instead of an int16t. It kind of needs to be an int16_t to support negative which is a valid thing. I think it would be ok to change that mavlink definition, but not really sure.

DonLakeFlyer, 2019-10-03:
@Hwurzburg Could you verify this is working now?

DonLakeFlyer, 2019-12-11:
@Hwurzburg Can you verify fix?

DonLakeFlyer, 2019-12-11:
Closing unless fix verification fails.

Hwurzburg, 2019-12-16:
tessted today with 12/13 daily build....positive throttle is fine...negative throttle throws very large positive numbers still

DonLakeFlyer, 2019-12-30:
@Hwurzburg Can you attach a telemetry log where you are moving the throttle back/forth that I can debug with?

Hwurzburg, 2019-12-31:
took a bit for it to arm, so wiggles are toward the end.... https://www.dropbox.com/s/nm2b3vuk5her1aa/2019-12-31%2000-10-06.tlog?dl=0

fyi, Mission Planner mavlink inspector shows it as always positive throttle in both directions....Mavproxy shows negatives as 653xx numbers....
The integrated OSDs in Ardupilot show + and - percentages properly
Tridge just made a  fix recently to VFR_HUD related to this also #13105

DonLakeFlyer, 2020-01-02:
I replayed the log and there aren't any values in VFR_HUD.throttle which are outside the 0-100 range. So as far as I can tell there isn't antyhig in the log which indicates a reverse throttle.

Hwurzburg, 2020-01-03:
I think I understand what is going on now:
The internal OSD display (for Rover only now) uses an internal actual throttle value which IS a signed number....and will show negative percents for reverse throttle....Mavproxy displays another MAVLINK variable which is 0-100 for positive throttle and 65535-65434? for negatives...

Tridge's PR fixes a different issue, that the internal OSD display doesnt not show reverse throttle in plane, because unlike Rover it does an ABS() function on the value before its sent to the internal OSD.... its fixed with his PR, but Mavlink HUD packets are still made to be only positive values before they are sent...that packet is defined as unsigned by MAVlink

so what I need to find out is what Mavlink info is being displayed in MAVPROXY for throttle percentage since it would allow the GCS to use it instead of HUD throttle to display negative percentages..its obviously being used by Mavproxy as unsigned when its really a signed value for throttle....will ask Tridge and Peter...thanks for hanging in there on this...

DonLakeFlyer, 2020-02-08:
> so what I need to find out is what Mavlink info is being displayed in MAVPROXY for throttle percentage since it would allow the GCS to use it instead of HUD throttle to display negative percentages.

@Hwurzburg Any update on this?