mavlink/qgroundcontrol #7855: Unable to use offline maps if default is deleted

seangrant82, 2019-10-03:
## Expected Behavior
When user deletes the Default tileset, it should remove the cache that is ther locally. this will help ensure that the offline maps you later download are coming from the qgctiledb and not in the cache

## Current Behavior
In Windows 3.5.4, if the user deletes the Default cache, they can no longer use offline maps. Rather than just the tilesets getting deleted, the entire row does whci prevents you from ever being able to import or add a  new set. You get he following error: 
[image]
Uninstalling and cleaning system of all temp files doesn't work.

## Steps to Reproduce:
Please provide an unambiguous set of steps to reproduce the current behavior
1. Open QGC and go to Offline maps
2. If there is a Default tileset, click the arrows and choose delete
In my case the entire default tiles was deleted and I am not able to use QGC offline

## System Information
When posting bug reports, include the following information
- Operating System: WIndows 10 1903
- QGC Version: 3.5.4

dogmaphobic, 2019-10-04:
I could not reproduce the error dialog you show above. What exact set of steps led to it? When you select the default tile set and select to delete it, you get this notice:

[image]

If I select "Yes", it proceeds to delete everything and I ended up with nothing as expected:

[image]

seangrant82, 2019-10-16:
That is exactly what i did. Is the default cache db stored in a local temp directory? I routinely purge my temp directories, would a combination of deleting the temp then deleting the Default cause this?

Also, I have unistalled and removed all directories related to QGC but whenever I install this or AGS, i still receive this error

Hasan-Ag, 2021-04-11:
Any solutions for this problem please

closed 2025-11-13 by HTRamsey