mavlink/qgroundcontrol #7857: Joystick values garbage when hotplugging joystick

julianoes, 2019-10-03:
First of all: thanks for the hotplugging ability, that's nice (almost)!

## Expected Behavior

If I unplug the joystick and plug it in again with QGC running I expect the joystick values/inputs to be sane and reflecting the actual joystick position.

## Current Behavior
However, what happens is that the joystick values go to extreme positions after plugging the joystick in again. They stay in extreme positions until you move a stick or press a button.

## Steps to Reproduce:

1. Configure/calibrate joystick and fly in SITL.
2. Unplug joystick while leaving QGC running.
3. Plug joystick in again: the quad yaws wildly, looking at the joystick tab the stick positions are in extremes.

## System Information
When posting bug reports, include the following information
- Operating System: Fedora 30
- QGC Version: daily 2f747d088 (2019-10-02)
- Flight Controller: PX4 SITL

## Log Files and Screenshots

This is after plugging in:
[image]

FYI @dogmaphobic @stefandunca

julianoes, 2019-10-03:
Actually, the same behaviour seems to happen when you open QGC with a joystick already plugged in.

DonLakeFlyer, 2019-10-03:
Joystick just seems completely broken. My sticks don't work at all.

DonLakeFlyer, 2019-10-03:
The stick positions on the general page don't update unless the joystick is enabled. The buttons do updates when joystick is disabled. This isn't the way it used to work. You used to be able to see what the joystick was doing no matter what. That way you could at least make sure it was working at all.

dogmaphobic, 2019-10-04:
That behavior was added in January last year:
https://github.com/mavlink/qgroundcontrol/commit/02a8b5b1d0fc9551f319553bd4f36e7257fdfbf8

Meaning, those messages are only emitted if the joystick is enabled. Should I remove the condition and let it emit all the time? The current implementation is:
`if (_activeVehicle->joystickEnabled() && !_calibrationMode && _calibrated) {`

DonLakeFlyer, 2019-10-04:
I vaguely remember something about that. Was it ArduSub folks who changed that? Might want to ask them why. There might have been some important reason then. If not, I think it's better to show the joystick values all the time

julianoes, 2019-10-14:
@jaxxzer your help would be appreciated. I was debugging this and made the following observations:
1. My Logitech Extreme 3D joystick doesn't show this behaviour, only the gamepad 'Logitech Logitech Dual Action' does.
2. When I reconnected the gamepad the [_update](https://github.com/mavlink/qgroundcontrol/blob/05db03a546b6c8d28fc209a46d86bc84c9ec3715/src/Joystick/JoystickSDL.cc#L158) function is called and grabs the axes values and the first 4 axes are all `-32768`. Once I press a button, they jump to 0 and are correct from then on.
3. The [JoystickSDK](https://github.com/mavlink/qgroundcontrol/blob/05db03a546b6c8d28fc209a46d86bc84c9ec3715/src/Joystick/JoystickSDL.cc#L77) gets created every time I plug the gamepad in again.

dogmaphobic, 2019-10-14:
I noticed the code makes a distinction when a "gamepad" is used (instead of a regular joystick). It makes some assumptions and uses some hardcoded values whereas for a "regular" joystick, you need to calibrate it before anything works.

jaxxzer, 2019-10-15:
> Logitech Logitech Dual Action

Can you let me know what product/model this corresponds to?

julianoes, 2019-10-15:
> Can you let me know what product/model this corresponds to?

Logitech Gamepad F310.

julianoes, 2019-11-20:
I just found that this is bug is only happening when the switch on the bottom of the Logitech Gamepad F310 is set to D. If it's set to X it works fine.

dogmaphobic, 2019-11-22:
This might need further attention. The Joystick code bypasses the normal flow if you attach a "gamepad". It skips calibration and uses hardcoded values. Maybe this thing is being recognized as a _gamepad_ but when the switch is set to "D" the values aren't right? I've never use a gamepad so I can't say what this is all about for sure.

dogmaphobic, 2019-12-26:
I don't have a "gamepad" to test this. @jaxxzer?

jaxxzer, 2020-01-06:
We've got the logitech f310 that @julianoes is using. We will check it here.

rizvanovic, 2022-10-12:
Still get issues like these. Not sure if it is the same exact issue but very similar to the behavior you describe.

zdanek, 2022-10-14:
I tried to reproduce it on
* ubuntu 20.04
* PS3 dualshock

it did not occure. I've even moved stick to edge position and disconnected it but it went back to center after reconnection.

I'd need more info from other users with respect to their OS and gamepad / joystick type to check if it's only single problem.
@rizvanovic which OS and gamepad do you have?

julianoes, 2024-03-04:
Haven't seen this in a while.

closed 2024-03-04 by julianoes