mavlink/qgroundcontrol #7858: Missing Parameters mentioned multiple times

MaEtUgR, 2019-10-03:
## Expected Behavior
If parameters that are necessary are missing from what the autopilot exposes you get a warning with the list of parameters that are missing each mentioned only once.

## Current Behavior
I get the warning but one parameter is mentioned 10 times in a row.

## Steps to Reproduce
Please provide an unambiguous set of steps to reproduce the current behavior
1. Load SITL or a vehicle with some parameters missing e.g. built from https://github.com/PX4/Firmware/commit/c322f1d1568b73327916ce2363c80375e3b5485b
2. Connect QGC daily to the vehicle or SITL
3. See warning with repeated first parameter

## System Information
When posting bug reports, include the following information
- Operating System: Found on Windows 10 and Android
- QGC build: latest daily
[image]

- Flight Controller: Pixhawk 4 mini, SITL
- Autopilot (with version): PX4 not exposing some commander parameters QGC is expecting

## Screenshot
[image]

MaEtUgR, 2019-10-06:
@DonLakeFlyer Thanks a lot for the fix! I tried to verify with my setup but couldn't because I ran into: https://github.com/mavlink/qgroundcontrol/issues/7878

MaEtUgR, 2019-10-08:
Verified to be fixed 👍 
[image]
Although I don't understand why it always says `-1` in front of the collon.

DonLakeFlyer, 2019-10-08:
Reopening to fix the -1. That is an internal QGC things which means default component id.

DonLakeFlyer, 2019-10-15:
-1 should be gone now as well

closed 2019-10-15 by DonLakeFlyer

MaEtUgR, 2019-10-16:
@DonLakeFlyer Now it shows 1 instead of -1. It's probably not what you expected because I still don't understand why:
[image]

DonLakeFlyer, 2019-10-16:
That is correct behavior. The 1 is the component id. Parameters with the same name could come from different components: PX4 Flow, ESP8266, ...

PR #7859 (2019-10-03): Remove duplicates from missing parameter reporting
Fix #7858