mavlink/qgroundcontrol #7864: Custom elevation data import

cosmarc, 2019-10-04:
Are there any plans to add the possibility of loading custom elevation data into QGC by loading 
 for example GeoTIFF, LAS, DEM, ASC files?

DonLakeFlyer, 2019-10-04:
Yes, someone is working their way towards that. But it is a large work item. Not sure when it will come in.

closed 2019-10-04 by DonLakeFlyer

tilaktilak, 2019-10-06:
I am on it at the moment, I expect sending a PR in couple of weeks for the second part of the job.

cosmarc, 2019-10-07:
@tilaktilak Would you like to share more about what formats you want to support and how did you envision the solution? We're trying to implement something similar and therefore we can probably share some work/ideas.

tilaktilak, 2019-10-07:
Oh sure,
my answer is here #7818 (to keep track of stuff at the same place).
I am definitively keen to share work/ideas with you guys :-) !