mavlink/qgroundcontrol #7865: MAVLink: Don't use source compid = 0 (MAV_COMP_ID_ALL) in message header

olliw42, 2019-10-04:
This is a follow up on https://github.com/mavlink/mavlink/issues/1244

## Expected Behavior
source component ID in MAVlink messages should be component ID of QGC, and not 0

## Current Behavior
QGC appears to (occasionally !?) send out MAVLink messages with source component ID = 0 in the message header, specifically e.g. the requestCameraInformation.

## Steps to Reproduce:
just run

## System Information
- OS: Win 7
- QGroundControl: this is for an "old" QGC version: QGroundControl Development HEAD:49ef8c141 2019-05-15 08:53:25-0700. Scanning through the recent PRs indicates that this behavior should exist also for the latest version.
- flight controller: irrelevant

## Detailed Description
As confirmed in https://github.com/mavlink/mavlink/issues/1244, any MAVLink component should send out the messages with the source component ID set to its own component ID. There thus can't be messages with msg.compid = 0. I now observed that if I add code to STorM32 which rejects all messages with msg.compid = 0 based on this expectation I find that QGC's requestCameraInformation command is rejected and the MAVLink camera service be broken. My debug output also explicitly shows that QGC's requestCameraInformation command has msg.compid = 0. 

There might be further messages send out by QGC for which this is the case, but I haven't tested or explored this.

To avoid confusion, this here is about sender id in the packet fields not target id in the payload.

dogmaphobic, 2019-10-04:
The camera manager requests the camera info and provides the _target_ component ID (the compID of the camera that just sent a heartbeat). The camera manager does not define the _source_ sysID or compID. That comes from `Vehicle`.

https://github.com/mavlink/qgroundcontrol/blob/master/src/Camera/QGCCameraManager.cc#L363

The `Vehicle` class in turn uses `_mavlink->getSystemId()` and `_mavlink->getComponentId()` respectively for source sysID and compID.

https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L3249

QGC has its compID hardcoded to 0:

https://github.com/mavlink/qgroundcontrol/blob/master/src/comm/MAVLinkProtocol.cc#L395

In other words, all commands sent from QGC will have the source compID set to 0.

olliw42, 2019-10-05:
many thx for the detailed outline of the flow of the code

I think what the MAVLink spec extended by the comments of @hamishwillee in https://github.com/mavlink/mavlink/issues/1244#issuecomment-538180227 is saying is that you must not use 0, but should use MAV_COMP_ID_MISSIONPLANNER = 190.

Hamiswille's words on using 0: "This is clearly a mis-use".

dogmaphobic, 2019-10-05:
@DonLakeFlyer Should I change that hardcoded 0 to `MAV_COMP_ID_MISSIONPLANNER`?

DonLakeFlyer, 2019-10-05:
MAV_COMP_ID_MISSIONPLANNER, news to me! Don't know. Considering it's been using 0 since before I became involved, I'd be concerned about changing it. You've got something like 10 years of history there. No idea how PX4 firmware will react.

Looking at Mission Planner code it does use MAV_COMP_ID_MISSIONPLANNER everywhere. But that still doesn't mean PX4 won't puke. 

@hamishwillee Do you know if firmware will be ok with this?

DonLakeFlyer, 2019-10-05:
I'll do some testing against SITL to make sure there are no propblems.

hamishwillee, 2019-10-06:
@DonLakeFlyer Yes, nothing should set itself as having system or component id 0 (though the spec currently only explicitly states this for system id, the same logic is directly applicable for components).
So if PX4 falls over we should change it. That would probably fall on @dogmaphobic, but if not I will look into it. 

More generally, I'd like to provide direction to recipients of such a message. I thing the cause of any problems resulting from this will not be obvious. So while I don't think this is going to be very serious my take is that a we should perhaps ASSERT a non zero sender id in the sending libraries, and recipients are allowed/expected to silently drop them. Reasonable?

DonLakeFlyer, 2019-10-07:
You can't assert (like c++ assert) anywhere in code related to flying a drone. That will crash out which is something you never want to happen.

hamishwillee, 2019-10-07:
In retrospect, obviously! Thank you for the fix.

closed 2019-10-07 by hamishwillee