mavlink/qgroundcontrol #7869: Kind reminder: CameraManager VideoStop still broken

olliw42, 2019-10-05:
the issue described in https://github.com/mavlink/qgroundcontrol/issues/7605 still exists in todays daily build

it unfortunately makes QGC of quite little use for folks wanting to use the camera manager

for your convenience I copy over the description and close the older issue:

When in video mode, pressing the red round button sends a startVideo() and the button goes square as usual, but the button gets GRAY and doesn't remain red. From then on it's not possible to get QGC emit a stopVideo() by clicking it. In photo mode it works.

olliw42, 2019-10-14:
@dogmaphobic @MatejFranceskin @hamishwillee 

I've added names which I believe I've seen before here in this context, sorry if I pinged you incorrectly

I'm confused.

The above bug is now existing for 3 month. It is a tiny bug, but it has the drastic consequence that the camera manager on the whole is essentially useless. 

I had now two explicit cases where folks wanted to use the camera manager but went away when I told them that they have to use a daily build of 3 months ago. This furthers my own feeling that it is very sad that QGC cannot be used for that purposes.

if possible, I really very much would appreciate feedback - even just one sentence - as regards to what's up here. 

Is it only and only me seeing this bug? Is the camera manager not used by anyone except of me? Is there no interest anymore in the camera manager from you side? ...? ...? ...?

thx in advance.

dogmaphobic, 2019-10-14:
Sorry but I've been out for most of the past several weeks and buried in other stuff in between. All of the above has been working fine in all devices/setups I have access to, which would indicate some edge case or configuration issue. With that said, I will see if I find a slot of time this week to go over your reports and try to see what could be going on. I have a suspicion the camera is not sending back camera status messages to QGC.

olliw42, 2019-10-14:
first, many thx for your feedback, really appreciated.

I'm using this on a Win7 computer, and I simply run the daily-installer, clicking yes everywhere, i.e also install the drivers etc.. 
It is not immediately obvious to me what I could do wrong.
Also, it was working fine before the QtQuick conversion (not sure I use the right words, but few months back you changed something like from QtWidgets to QtQuick or so, before that it was all working fine, after that it stopped working fine).

The camera is sending back camera status messages. I can see that also in QGC when I enable the respective log. And as said, when I use the "old" daily build of 5 months ago it's all working fine. So, I would rule out that possibility, and effectively would rule out any such extrinsic issues, but believe it's inside QGC.

(what I would need to explicitly test again is if the latest daily build does send out requests, I think I have seen that but am not sure to 100%, however the camera is sending back if asked for)

I just made a quick screenshot for what I see with the "old" daily build (this "old" daily I use as a kind of reference since it's the last I found working)(for a test with the latest daily I first would have to install it)
for a very simple camera:
[image]
and here the same for a more sophisticated camera using also camdefs etc.:
[image]

olliw42, 2019-10-14:
so, just installed the daily of just now. 

With the very same setup and settings as for the last screenshot in the previous post I see for the log:
[image]
so the CaptureStatus messages are emitted and received

this is a screen shot of the bug:
[image]
after I pressed start video the first time, the button goes gray, and stays gray, and then I press it again nothing appears to happen, especially the video is not stopping

hope this may help
thx

olliw42, 2019-11-04:
resolved
hence I close it 
:)

closed 2019-11-04 by olliw42