mavlink/qgroundcontrol #7874: Tag for last qt widget build is screwing up versioning of Play android

DonLakeFlyer, 2019-10-05:
Version is displayed as the lastQtWidget tag instead of a version number

DonLakeFlyer, 2019-10-05:
@dogmaphobic Do you think you could changes this to a branch and remove to tag?

dogmaphobic, 2019-10-06:
I did that way back when. The branch is here:
https://github.com/mavlink/qgroundcontrol/tree/LastQtWidgetBuild

I had also deleted the tag but it seems it's "stuck". I just tried deleting the tag again and even though it seems it goes through fine, when listing the tags, it is still there.

DonLakeFlyer, 2019-10-06:
Thanks Gus. I'll try to fight with it then.

DonLakeFlyer, 2019-10-06:
Hmm, it doesn't show up here any more: 
[image]

Confused...

DonLakeFlyer, 2019-10-09:
Looks to be all good now. On to the next problem with Google Play...

closed 2019-10-09 by DonLakeFlyer