mavlink/qgroundcontrol #7875: White screen on flight screen in Android 8.0 phone

Hwurzburg, 2019-10-05:
beginning with the first daily builds in Oct or so, on start up the flight screen shows the map fine with QGC gps location....after connecting and loading params, switching to fly screen is white in all the map areas....planner screen is still fine

sometimes, if I switch from connections screen quickly after params start downloading, the fly screen will be okay with map tiles shown

also, I have noticed that mode changes fail often under these conditions....

as of now, the daily builds on my Motorola RAZR Android 8.0 phone are not useable...

DonLakeFlyer, 2019-10-06:
@Hwurzburg Can you look at Settings/Console to see if there is any interesting log output in there when this happens.

Hwurzburg, 2019-10-06:
okay...white screen does not happen when using OTG USB to connect to vehicle....only when using telem (actually I use a 3DR Sik radio to BT repeater and then connect QGC via the BT)....
console shows errors like :"Elevation tile fetching returned error(QnetworkReply:operation cancelled error) and TLS initiailization failed

if this is of any help

closed 2026-10-04 by github-actions[bot]