mavlink/qgroundcontrol #7877: Heading information in Follow mode not being updated to a useful accuracy

Hwurzburg, 2019-10-06:
The heading info being broadcast in GLOBAL_INIT from an Android phone is not accurate enough to be useful.....when walking, it produces almost continuous 0deg values....if you move fast enough, it occasionally reports a heading approximating the movement direction but quickly reverts to 0deg...
perhaps the phone GPS should be put into Pedestrian mode (Navigation apps in walking direction mode produce fairly good directional icons for current heading of the phone), or the use of the internal compass be used for heading...this would require that QGC not flip its display...since its already in landscape mode, only two orientations are possible, so fixing it to one should be acceptable so that heading is always intuitive. Otherwise, the Follow Me panel is not going to be useful.

DonLakeFlyer, 2019-10-07:
I need to switch to using the Sensor information for Compass is available.

DonLakeFlyer, 2019-10-09:
I have this working with the Compass now but seem to be running into a Qt Android bug which reporting screen orientations.

Hwurzburg, 2019-10-09:
cool...let me know when you want me to dl a new daily build and try it out on the boat...

DonLakeFlyer, 2019-12-25:
Tried fighing with this again. I still can't seem to get good data from the device in order to use Compass for heading. Will have to try again later.